SeedSigner/seedsigner #1: Create online build instructions (without needing a 2nd Pi)

mflaxman, 2020-12-15:
I was able to get something like this working on a different project, I need to modify it for this one and then add to readme if it works.

#### apt-get install replacement

Online:
```bash
$ mkdir apt && cd apt
# wget all packages from https://packages.debian.org
```

Offline (after copying files over):
```
$ sudo dpkg -i *.deb
```

#### pip install replacement

Online:
```bash
$ mkdir pip-whl && cd pip-whl && python3 -m pip download pillow numpy RPi.GPIO spidev pip setuptools && wget https://www.piwheels.org/simple/numpy/numpy-1.19.4-cp37-cp37m-linux_armv6l.whl && wget https://www.piwheels.org/simple/pillow/Pillow-8.0.1-cp37-cp37m-linux_armv6l.whl && cd ..
```

Offline (after copying files over):
```bash
$ sudo python3 /path/to/pip-20.2.4-py2.py3-none-any.whl/pip install --no-index --find-links [everything in pip-whl]
```

If this works, then I think all these `deb`/`whl` could be brought into the project some how to make checkout + install easy. I'm not sure if the best approach is git submodules or perhaps an installer script to fetch them and validate their digest.

mflaxman, 2020-12-15:
Also, here are instructions on how to ssh into Pi Zero via USB (no internet):
https://desertbot.io/blog/ssh-into-pi-zero-over-usb

SeedSigner, 2020-12-15:
Appreciate these thoughts. Good info on the ssh via USB and the installer script is an interesting idea. How much complication would there be in simply posting a Raspberry Pi OS Lite image with everything preconfigured?

mflaxman, 2020-12-16:
Pretty straightforward and I think a good idea! Obviously better if people build from source, but I'm sure some people (who are using this only for redundant secondary verification) will be lazy.

mflaxman, 2020-12-16:
Instructions here btw: https://thepihut.com/blogs/raspberry-pi-tutorials/17789160-backing-up-and-restoring-your-raspberry-pis-sd-card

Making this deterministic for trust-minimizatino would be harder, but if it's just for redundant verification anyway that'd be OK (at least the Pi Zero without wireless has a solid airgap).

SeedSigner, 2020-12-17:
Realized a 4GB microsd would be the best option for creating an image; going to acquire one then will compile a clean installation and post the image.

SeedSigner, 2021-03-31:
Packaged release into burnable .img starting with version 0.1.0, thanks again for the suggestions!

closed 2021-03-31 by SeedSigner