SeedSigner/seedsigner #5: Please use a GPG key to sign releases and commits.

jonathancross, 2021-01-20:
In order to reduce trust in GitHub and increase reliability of the project / prevent forgeries, please generate a GPG key and use it when making commits or at the very least add a signature when making a release.

This is standard practice for Bitcoin software, especially wallets.

Thanks!

SeedSigner, 2021-05-19:
Great suggestion, this has been implemented for releases.

closed 2021-05-19 by SeedSigner

jonathancross, 2021-05-21:
I see the latest release now has a gpg file, but it is unclear which data this is signing.
Normally, I'd expect a file named  `seedsigner_0_4_0_b3.txt.sig` which was a detached signature of `seedsigner_0_4_0_b3.txt`.

```
gpg --verify seedsigner_0_4_0_b3.txt.gpg  seedsigner_0_4_0_b3.txt
gpg: not a detached signature
```