SeedSigner/seedsigner #14: Selecting third character hangs

jcp-os, 2021-05-20:
Freezes when selecting a third character on seed word search. left back not working, then hangs and have to reboot.

SeedSigner, 2021-05-20:
Am curious about the variables around this -- had other seeds already been entered, or other operations been done before this, or was it the first seed entry attempt after boot, etc. etc. If related to programmatic structure (not hardware), the structure will have changed significantly moving into 0.4.0. Going to keep this open for now and make sure to thoroughly test this under the new structure, but current changes may obviate the need for further investigation. Really appreciate you coming here with it.

jcp-os, 2021-05-20:
Really appreciate the work! This tool is awesome.

Here's how to reproduce the bug:

1. Boot > Seed Tools > Store a seed
2. Use seed slot #1
3. Use 12 word seed
4. right > a > right > a > left > "then hangs"

Usually happens on 2nd to 3rd seed selection.

newtonick, 2021-06-01:
I'm unable to reproduce this bug on version 0.4.0

SeedSigner, 2021-06-10:
Thank you again for submitting this! This workflow is no longer possible after recent updates, as only valid seed letters will appear during the word entry process.

closed 2021-06-10 by SeedSigner