SeedSigner/seedsigner #17: Add support for Accounts in the derivation path

newtonick, 2021-06-11:
Derivation path in seedsigner currently only supports m/84'/0'/0'/2'. Would be a good advanced feature if the derivation path could be altered specifically to add account support. If the "account number" could be incremented in an advanced settings screen (m/84'/0'/{account}') this would allow for multiple wallets with the same seed phrases.

SeedSigner, 2021-06-23:
Agree that this would be a very cool and useful feature to implement. The trick with the settings in SeedSigner is going to be getting the menu settings so they are not simple. It also calls into question some sort of persistence for settings, as we add more options in settings. But as to this particular account number option, agree that it would be a great advanced feature.

newtonick, 2021-08-06:
moving to [PSBTParser](https://github.com/SeedSigner/seedsigner/blob/ab68f2b9a384de30252df0a6738ea66316b15b1a/src/seedsigner/models/psbt_parser.py) has resolved this issue.

newtonick, 2021-08-06:
Current plan is to add custom derivation support, not just account number for bip84

TKone7, 2021-08-23:
I am very new to SeedSigner (actually digged deeper through the settings for the first time yesterday). And this was the first shortcoming I noticed. Wanted to import a (dummy) singlesig wallet into Specter and had to go the detour via Electrum. Maybe there would have been an easier way (please let me know). But I would very much support the option to export a specific account over QR. 

Would also be up helping implement / test this.

newtonick, 2021-12-02:
Custom derivation path now allows accounts. Closing issue

closed 2021-12-02 by newtonick