SeedSigner/seedsigner #21: Fast Automatic Seed Words Input

carlosmassa, 2021-06-20:
It would be nice to add a feature that allows for easy and fast input of the 12/24 seed words. For instance, by using the camera to scan a paper with the written words or its BIP39 corresponding number and applying OCR or a template made-for-purpose that would make the recognition easier. This way the user can keep the seed stored safely and once needed to sign, he or she would just present it in front of the camera for automatic input.

This could be an option, so users could choose to enter the words manually as it is done today, or via this new, fast, automatic method.

SeedSigner, 2021-06-23:
I agree that some sort of optical scanning feature to input seed words would be a great feature. At first we'd thought of attaching a small offline thermal printer to seedsigner to print QR codes with the seedwords encoded into them, but that seems like a clumsy/elaborate solution. I think being able to attach a thumb drive to SeedSigner and save the words to the thumb drive might be a better solution. Deciding to encrypt on the thumb drive and how to execute that is another challenge; people store seed words in plain text so as long as the users knew the seed was being stored in plain text on the thumb drive, it could be an acceptable first iteration of a solution for easier inputting of seed words. As of right now, SeedSigner is targeted at savers doing relatively few spends, so creating a mechanism for super-fast seed input as been less important. I am open to any other thoughts on this as well.

kdmukai, 2021-06-25:
My take on the idea: #29

kdmukai, 2021-06-25:
[image]

Took me about 20min to transcribe a 12-word seed by hand. Not terrible. Maybe practical enough?

carlosmassa, 2021-06-25:
The QR code is a very practical idea. However, the point of the new feature would be to save time, and although 20 minutes is not that bad to create the QR code manually, it is prone to manual errors and you still have to put in the time.

What about printing the QR code directly from the Seed Signer, through a usb connection to a small and cheap thermal printer. Something like this: https://www.amazon.com/dp/B06VXPLLMQ/ref=cm_sw_r_cp_api_glt_fabc_CBE0S2K5W16FPSTXNPMM

kdmukai, 2021-06-25:
@carlosmassa my main use case is thinking about folks in places like El Salvador where even a $50 SeedSigner would be considered an expensive item.

I also think it would be good to create a printable template of a blank QR code that has a grid and the basic structural elements (e.g. the concentric boxes in three of the corners) already there. That'll make it a bit easier to fill out. This assumes access to a printer, of course. But I could imagine training sessions at places like Bitcoin Beach where they can bring in lots of people, train them up, provide printed out QR code templates for them to take back home, etc.

It's looking like I might be able to simplify the QR code further and get it down to 21x21. That would make it significantly easier for a human to transcribe.

SeedSigner, 2021-06-26:
While the idea of a thermal printer would be cleaner solution, I have always had reservations about the idea of buying a separate device that you're only going to use with your SeedSigner once, or at most a handful of times. Also a good point that people of less means may not necessarily have the additional funds to purchase a secondary device.

I would also note that even without the printer or the manually-generated QRs, a user can always write down 24 words and reasonably easily/quickly enter them back into SeedSigner; that option already exists. The point of the QR is to be able to quickly / more easily import a seed, to make SeedSigner more convenient for perhaps slightly more frequent transactions. I think it seems reasonable to manually color a QR for 10-20 mins to have that perpetual convenience.

The thermal printer is a really clean solution but I lean towards the manually generated QR for cost considerations.

bitcoinheiro, 2021-06-29:
Very nice, just saw your demo @kdmukai
Users need to be extra careful with these backups since they're quite easy to be read by any machine with a camera.
I would recommend users to keep this code concealed with a good cover and only expose it in front of the seedsigner - and then cover it imediatelly back after scan.

carlosmassa, 2021-06-29:
All great points @kdmukai and @SeedSigner. I also saw the demo video from @kdmukai and it does look like a great solution. Great improvement!

SeedSigner, 2021-07-26:
I have used this feature myself several times, great stuff, much faster seed entry has arrived for SeedSigner.

closed 2021-07-26 by SeedSigner