SeedSigner/seedsigner #23: Bug signing 1-of-2 multisig transaction when recipient is same wallet

kdmukai, 2021-06-23:
Both keys are SeedSigner testnet devices in Specter.

Transaction is sending coins back to a new addr in the same wallet.

SeedSigner is hanging on "Parsing PSBT"
```
Traceback (most recent call last):
  File "main.py", line 12, in <module>
    controller.start(DEBUG)
  File "/home/pi/seedsigner/controller.py", line 69, in start
    self.show_main_menu()
  File "/home/pi/seedsigner/controller.py", line 91, in show_main_menu
    ret_val = self.show_sign_transaction()
  File "/home/pi/seedsigner/controller.py", line 339, in show_sign_transaction
    signer.parse_psbt(raw_pbst)
  File "./helpers/psbt_signer.py", line 76, in parse_psbt
    (self.change, self.fee, self.spend, self.destinationaddress) = PSBTSigner.change_fee_spend_amounts(self.tx, self.inp_amount, policy, self.seed_storage.get_network())
  File "./helpers/psbt_signer.py", line 260, in change_fee_spend_amounts
    return (change, fee, spend, destinationaddress)
UnboundLocalError: local variable 'destinationaddress' referenced before assignment
```

I can confirm that the same device/seed works fine signing a tx for an address in a different wallet. So it's just the self-send/cycling coins that seems to cause problems.

newtonick, 2021-06-30:
@kdmukai do you remember which branch or release you found this issue on?

newtonick, 2021-08-06:
moving to [PSBTParser](https://github.com/SeedSigner/seedsigner/blob/ab68f2b9a384de30252df0a6738ea66316b15b1a/src/seedsigner/models/psbt_parser.py) has resolved this issue.

SeedSigner, 2021-08-06:
Going to wait until the change gets worked into a release, then will close this.

newtonick, 2021-12-02:
@SeedSigner I believe this issue can be closed

SeedSigner, 2021-12-04:
Thank you for the reminder.

closed 2021-12-04 by SeedSigner