SeedSigner/seedsigner #24: R&D: Build SD card image within Docker raspi VM?

kdmukai, 2021-06-23:
Not sure if this is possible but being able to set up the SD card within a Docker VM would probably go a lot faster than doing it onboard a Pi.

The whole process could then be scripted via a `Dockerfile`. Then you wouldn't need an internet-connected Pi to build on (nor the internet access over USB option). You'd basically just run `docker-compose build` and scoop the results out to the SD card. This scriptability would also make it easy to make alternate versions of the software (e.g. a build that defaults to Spanish, a debugging build, a production build, etc).

I'm just not sure how you'd get the Docker image back out to SD and whether it would be compatible with an actual Pi (I suspect/hope this is all straightforward, though).

Here's a typical `Dockerfile` for my python projects:
```
# Dockerfile

# Pull base image (lambCI images mimic AWS Lambda)
FROM lambci/lambda:build-python2.7

# Set environment variables
# Don't write *.pyc files or create __pycache__ dirs
ENV PYTHONDONTWRITEBYTECODE 1

# Stream python stdout back to calling shell
ENV PYTHONUNBUFFERED 1

# Install additional required tools
#               nano: sane text editor
#                 nc: netcat required for `wait-for` script
#   postgresql-devel: required for psycopg2-binary
#        mysql-devel: required for python 'pattern' dependencies
RUN yum update -y
RUN yum install -y \
    nano \
    nc \
    postgresql96-devel \
    mysql-devel \
    ruby23 \
    ruby23-devel

RUN gem install pg

# Set work directory
WORKDIR /code

# Set up the interactive bash shell to our liking
COPY docker/bashrc /root/.bashrc

# Create virtualenv and install dependencies
RUN virtualenv ~/.env
COPY requirements.txt .
RUN ~/.env/bin/pip install -r requirements.txt --no-cache-dir
```

see: https://github.com/lukechilds/dockerpi

SeedSigner, 2021-06-23:
This is something to consider as we move forward, the move to compiling for tinycore was supposed to dramatically reduce the size of the microsd burn (and the size of mem card required) but things are moving very quickly and so much to do.

newtonick, 2023-08-26:
SeedSigner OS replaces the need for this idea.

closed 2023-08-26 by newtonick