SeedSigner/seedsigner #32: Grind for low R signatures

craigraw, 2021-06-28:
(This issue originated at https://github.com/sparrowwallet/sparrow/issues/144 - transferring it here)

SeedSigner does not appear to be currently grinding for low R signatures, resulting in a larger transaction size than is necessary. In particular, look at the signatures for `Input #2` of the PSBT attached - when encoded to DER, both signatures are 71 bytes in length. Signatures with low R are less than 71 bytes in length. It's strongly recommended to grind for low R signatures - this is done by adding additional data to the deterministic nonce as described by section 3.6 of RFC 6979, and signing until a signature with low R has been calculated. Bitcoin Core, Electrum, Sparrow and other wallets do this.

Further, it's also important to calculate this nonce the same way as Bitcoin Core/libsecp256k1 (which results in the same signature). Since we can't easily prove that the nonce is random, it is strongly advantageous to use the same nonce as everyone else. Here are some notes that may be helpful:

1. The initial attempt to sign uses no additional data as per RFC 6979 3.6. If that attempt results in high R, the second attempt uses a value of 1 (and further attempts increment).
2. The additional data is encoded to little endian.
3. The additional data is applied in both RFC 6979 steps 3.2d and 3.2f (RFC 6979 3.6 does not clarify this well).

It looks like actually signing is being done by the embit library [here](https://github.com/diybitcoinhardware/embit/blob/5b72733b4fe7c74cc485f02c46484a9e791c0607/src/embit/psbt.py#L658) - but posting this as a SeedSigner issue because I think it's an important one to address given the focus of this project.

[test.consolidation.fee.error.if.left.on.lowest.range.psbt.txt](https://github.com/SeedSigner/seedsigner/files/6725229/test.consolidation.fee.error.if.left.on.lowest.range.psbt.txt)

SeedSigner, 2021-07-07:
A relevant change has been made in embit that should resolve this (https://github.com/diybitcoinhardware/embit/issues/12).

newtonick, 2021-07-08:
I've done testing with embit 0.4.3 release and can not replicate the issue found before. This appears to be resolved.

SeedSigner, 2021-07-08:
Going to leave this open until we get the updated version of embit into a release, thank you for testing the fix.

SeedSigner, 2021-07-26:
Was able to circle back and test this, have done several transactions with a large number of inputs, haven't encountered the fee issue again, considering this closed.

closed 2021-07-26 by SeedSigner