SeedSigner/seedsigner #35: SeedSigner GUI

bc1easy, 2021-06-30:
[image]

Create a user-friendly GUI to make the SeedSigner software interface more accessible to non-technical users.

### Design Goals:

**Visual Design:** Improve aesthetics and provide visual aids using UI treatments and iconography.
**Interaction Design:** Increase overall efficiency of user inputs and feedback.
**Information Architecture:** Improve findability by organizing and surfacing common tasks.
**Content Design:** Simplify language to make abstract concepts more accessible to the average user.

### Proposed Enhancements:

* **Navigation:** Add Navigation Bar for context of location and easier access to global navigation functions
* **IA:** Accessible groupings of features, surface high-traffic actions
* **Content:** Simplify labels for viewport optimization and localization
* **Visual:** Add iconography to reduce cognitive load and aid localization
* **Visual:** Change font (Impact) to a more legible but space-efficient typeface

### Other Enhancements:

* Pagination system for fitting >4 menu items on screen
* 'Appearance' settings screen for UI color customization
* 'About/Info' settings screen for version/system information

SeedSigner, 2021-08-16:
I think we're getting closer to starting to implement this vision, need to have some more offline discussions.

kdmukai, 2022-12-07:
> Does this deserve to be closed now?

Yes, I think we accomplished it!

closed 2022-12-09 by newtonick