SeedSigner/seedsigner #68: [Feature] Display Seed Binary

jakob6102, 2021-08-12:
Add the option to view the binary of your seed and input your seed as binary. Displayed as 12/24 11 digit strings representing the binary for each word.

This would expand the options for seed storage and make metal backups like this (https://github.com/jakob6102/seedcard.git) quicker, easier and safer to create. It also has benefits when inputting your seed to a seedsigner, binary only needs 11 button presses per word if say Key 1 is used for 1 and Key 2 is used for 0.

SeedSigner, 2021-08-16:
Like the concept of this idea. Going to keep this open, curious how many people are using this kind of backup scheme.

CODLCO, 2024-04-10:
I was going to suggest a very similar feature but instead of displaying it as binary, also display it as decimal. 

This allows for a more compact backup and would help all of our customers who use seedsigners.

The binary entropy for 12 words is chunked into 11 pieces then converted into decimal, and that is what is mapped to the word list and displayed so this is one less step logic wise on both the backup and restore.

nosedam, 2024-08-28:
Hi guys! I really like this idea, I'm implementing this if no one is already working on this.

I have the backup options already done for binary and decimal, and I'm working on the restore functionality before making a PR if thas okay