SeedSigner/seedsigner #69: Bias in dice based entropy

crwatkins, 2021-08-15:
I believe there is a bias in the **Generate Seed with Dice** option.  I believe the code is using the dice to create a base 6 number and then simply using the (base 2) bits to create the entropy to be used for the seed.  This results in an uneven distribution in the result.    I explain some methods to avoid this [here](https://github.com/iancoleman/bip39/issues/435#issue-687469526).

In a practical sense, as long as the rest of the implementation is correct, the existing algorithm yields 99 unbiased bits given the required 99 rolls of the dice as explained [here](https://github.com/iancoleman/bip39/issues/435#issuecomment-690631830).

If one is happy with 99 unbiased bits of entropy, plus many more biased bits (which I suspect most people would be) then perhaps this is just fine.   However, in the reference above, I do explain some options to provide more unbiased entropy using the same number of dice rolls.  The engineer in me tends to try to avoid any bias at all.

In an unrelated issue affecting the same option, it looks like the last (mnemonic) word is discarded when creating the correct checksum.   This throws away three bits of entropy (because the last word represents three  bits of entropy plus eight bits of checksum).   It's hard to imagine that these three bits of entropy would have a significant practical affect on the security of the keys, but it is really just as easy not to throw them away and thus not have to think about that.

SeedSigner, 2021-08-16:
Relative to the latter issue, it's been raised before and is a known limitation of the embit library. The maintainer of that library is of the opinion that the discarded three bits of entropy aren't meaningful so no change has been made that would allow users to select a checksum word from the available options.

crwatkins, 2021-08-17:
> Relative to the dice entropy conversion issue, the other devs and I would welcome PR with proposed revisions to the code if there is a way to implement the feature that will make more sense. Thank you for your interest in the project!

Thank you for a great project!   Since I review many wallets, I (really) regret that I don't have the bandwidth to contribute to individual projects for the issues that I find.  (To be clear here, I wasn't formally reviewing SeedSigner, I just happened to notice that it requested 99 rolls of dice and that is a big red flag to me indicating some invalid assumptions regarding the generation of 256 bit keys.)   I would recommend referencing [BlueWallet](https://github.com/BlueWallet/BlueWallet) and [BIP39 Tool](https://github.com/iancoleman/bip39) for entropy generation methods with dice.

jpph, 2021-08-23:
You can have a look of Ian Coleman bip39 dice implementation. Having a reference is nice to show people they can test them self that same dice roll achieve same seed on Ian's page and your.

SeedSigner, 2021-10-13:
Going to go ahead and close this because we are moving towards mimicking the ColdCard dice implementation. Again, really appreciate the input.