SeedSigner/seedsigner #71: [Feature] QRcode option for passphrase input and backup

openoms, 2021-08-15:
Being able to scan the seed from a QR code is a great feature, but the manual input of the passphrase discourages the use of long phrases.
An option to show the passphrase as a QR code would make it possible to have it saved the same way the seed can be stored and an option to scan it back would speed up the loading of the device.

The complication is that the character types and the length of the passphrase is not fixed so cannot be translated to just numbers from the BIP39 wordlist.
In case of for example using another 12-24 words as the passphrase would make the QRcode quite large. This limits the ability to record the QRcode manually more than the scanning of it.

Still I think the option would encourage the usage of long passphrases and improve the security of the keys.
A long passphrase is also likely a better (or at least an alternative) solution to an encrypted QRcode.
Both the seed and the passphrase itself can act as a decoy wallet and provide plausible deniability.

SeedSigner, 2021-08-16:
Have discussed this a little bit with Nick and we are still mulling over how it would make sense to implement. Appreciate the suggestion.

kdmukai, 2021-09-06:
A 25-char passphrase should fit in a 21x21 QR, no? That's not bad. And then a 25x25 would fit up to 47 chars.

[image]

My bigger concern is that you'd be reducing your 2nd factor of security to be essentially the same as your 1st factor. I definitely dislike the pain of typing in a long passphrase but I really like that I can store that passphrase in a totally different manner from the seed QRs which I need to have physically secured yet still accessible (bad combo of traits, obv).

david-bakin, 2021-11-25:
With the help of some conversation on Telegram I arrived at the following suggestion:

1. If you want 2-part security but only want to hide 1 QR code around the house and are ok with memorizing a short memorable passphrase then: use a passphrase with the current UI for entering it (or something like it).  And you get decoy wallets by using the QR code with _no_ passphrase or with any different decoy passphrase.
 
1. If you want the ease of entering 2 QR codes for two-part security, and are ok with hiding two separate QR codes around the house (in whatever plausible manner you like) then you use seedxor.  And you _also_ get decoy wallets by using either one or the other of the QR codes or any _other_ QR code you've also hidden around the house (easier to find, heh..., when the knowledgable attacker finds 2 QR codes he ... _stops looking for more!_)

1. And finally, if you decide to convert _from_ seed phrase + passphrase _to_ seedxor:  SeedSigner can help you there by giving converting your BIP39-wallet to a BIP-32 wallet - which is what Coinkite calls ["lock down the seed"](https://coldcard.com/docs/passphrase#related-feature-lock-down-seed)!

openoms, 2021-11-27:
@david-bakin great recap thanks.

Agree with your point 1 (similar to what @kdmukai wrote above), although being able to type passphrases should not stop the UI having and option to scan them as well, but I understand it is not a priority.

re 2: being able to add two seeds together with seedxor would add the same security as seed + PP and being discussed here: https://github.com/SeedSigner/seedsigner/issues/43. It is a very good option for new or seed-only wallets.

re 3: @david-bakin not sure how "converting a BIP39-wallet to a BIP-32" would help here.  Seedxor can only generate an other seed and we can't assign a a new seed to the wallet which has it's private keys generated from words+passphrase.

david-bakin, 2021-11-27:
ah, my mistake - I didn't know the seed wasn't the xprv itself. so NVM about that.

KyleOfTheCorn, 2023-01-24:
Bumping this. While working with the [Seed Tool](https://github.com/BitcoinQnA/seedtool), I've found that it would actually be very beneficial to users, especially with airgapped computers, to be able to type in their passphrases on a real keyboard, then scan them in with the SeedSigner. Which ever standard is used to display the QR code could then be implemented in Seed Tool.

CC @SuperPhatArrow & @BitcoinQnA for visibility.

dmonakhov, 2023-11-14:
If we are talking about use case I have one in mind, it is related with cold backup vs normal-use

- backup : If seed is used without passprase is makes backup less secure/private,  so it would be good to  store seed replicated  in one set of locations (friends in one country), and pass phrase in other set of locations( another country). In that case I do not have to make steel plate procedures, just many copies of papers should guarantee that at least one backup does exist in each of groups. 
- normal-use:  I trust my location so I want to store seed+passprase  as one peace and have an QR code scan easy experience.
So right now I have to chose, to have backup with non ideal privacy, or to have a pain of typing passphrase every time.

earthdiver, 2024-04-10:
I am interested in having a QR code option for entering CKJ passphrases. Although the software wallets I use (Sparrow, Electrum, BlueWallet, etc.) accept CKJ passphrases, I am unable to use them with SS due to its lack of support for non-ASCII passphrases.
I prefer the 29x29 QR code format, which is capable of storing 53 bytes of UTF-8 data.

kdmukai, 2024-04-10:
For those of us (me!) who are unfamiliar with the term:

https://en.wikipedia.org/wiki/CJK_characters

earthdiver, 2024-04-10:
@kdmukai thank you for your supplementary comment.

I don't think we need to fully understand the encoding. We simply need to be able to directly read the byte array data as the salt for the PBKDF2 function from the QR code.

Personally, I don't mind if the passphrase displayed for verification on SS gets garbled due to the ASCII font. By verifying the fingerprint, you can confirm if it has been read correctly.

earthdiver, 2024-07-31:
It didn't seem to be an issue that interested many people, so I took care of it myself over the weekend (input only).
Integrating with smartphone apps like 'Binary Eye' makes it easier to input long passphrases.
https://github.com/earthdiver/seedsigner

lorenzyannick, 2025-04-13:
Will this feature be implemented ?

kdmukai, 2025-05-31:
> If we are talking about use case I have one in mind, it is related with cold backup vs normal-use
>
> * backup : If seed is used without passprase is makes backup less secure/private,  so it would be [...]

Single sig + bip39 passphrase is not a great deep cold storage option because, as you describe above, you'd have to travel to country A to collect a copy of your seed phrase and then BRING it to country B so you have it on hand when you enter your bip39 passphrase. At that point you're at max vulnerability as all secrets are at the same place for that moment in time. SeedXOR and Shamir strategies have the same problem.

With multisig you can just go to country A, sign the tx, then take that partially signed tx to country B and sign with the second key. The keys themselves don't have to move or ever be brought together. You could obviously also have remote cosigners (people you trust to have one of your multisig keys) that can operate in complete isolation.

> * normal-use:  I trust my location so I want to store seed+passprase  as one peace and have an QR code scan easy experience.

If you need more frequent / convenient access to make onchain spends, I'd just create a completely separate spending wallet that holds a smaller amount of funds. The spending wallet can have a different set of security vs convenience tradeoffs, as you see fit.

> So right now I have to chose, to have backup with non ideal privacy, or to have a pain of typing passphrase every time. But I want both advantages, secure and private backup (seed+passprase), and easy to use seed+passprase as QR code. What do you think?

The SeedSigner device as a whole encourages users to experiment and create lots of seeds. Break out of the ONE PRECIOUS SEED mindset. Create the setups that make sense for each use case. Trying to blend them all together into one solution just compromises the priorities of each individual use case.

flmmkr1, 2026-08-17:
I would like this along with encrypted SeedQR support. Krux has it, and CryptoGuide/3rditeration added to his SeedSigner fork. Please bring those features over to the official version.

Wolf-of-Wal, 2026-09-04:
Hi — I'd like to add support for the passphrase scanning/input part of this long-standing feature request.

With hardware-wallet vulnerabilities, implementation mistakes and vendor-trust concerns continuing to remind Bitcoin users how important robust self-custody is, I think this feature has become even more valuable than when #71 was originally opened.

One of SeedSigner's great strengths is precisely what it doesn't require: no permanent key storage, no network connection, and a deliberately constrained camera/QR air-gapped communication model. For security-conscious users who want to use a strong, longer BIP39 passphrase, however, entering that passphrase repeatedly through the current on-device keyboard can become a significant usability obstacle.

That creates an unfortunate tension: the users most motivated to strengthen their passphrase security are also the users who encounter the greatest input burden.

Would the maintainers therefore consider a narrowly scoped “Scan Passphrase QR” option inside the existing BIP39 passphrase flow, while retaining manual entry exactly as it is today?

A minimal implementation could simply accept a standard QR code whose payload is the exact UTF-8 passphrase string, only while inside the BIP39 passphrase-entry context, and then hand that decoded string to SeedSigner's existing passphrase handling, review and fingerprint-confirmation workflow.

In other words:

BIP39 Passphrase → Enter Manually / Scan QR → existing Review → existing Fingerprint confirmation

I don't think SeedSigner necessarily needs to display or export passphrases as QR codes for this to deliver most of the benefit. Scan-only support would be enough, and would keep the change smaller and the security implications easier to review.

Ideally:

- manual passphrase entry remains fully available;
- QR scanning exists only inside the passphrase-entry context;
- the QR payload is passed into the existing passphrase-processing path without silently modifying the user's input;
- spaces, long inputs and Unicode are handled correctly;
- the existing passphrase review and fingerprint confirmation remain mandatory;
- no USB, networking, Bluetooth or other additional interface needs to be enabled;
- malformed, empty, long and Unicode inputs are covered by tests.

To me, this feels particularly well aligned with SeedSigner's philosophy: use the communication channel the device already trusts — its camera — rather than introducing another interface simply to make strong passphrases practical to use.

At a time when many Bitcoiners are thinking more seriously about hardware trust and resilient self-custody, this relatively focused feature could make a meaningful difference. It could allow users to increase passphrase strength without paying such a large usability penalty, while preserving the stateless, air-gapped design that makes SeedSigner so compelling in the first place.

We love what SeedSigner stands for, and this isn't a request to trade security for convenience. Quite the opposite: it is a request to make stronger self-custody easier to practise without weakening SeedSigner's existing security model.

Is this something the maintainers would consider suitable for upstream SeedSigner?

Thanks for all the work that goes into the project.