SeedSigner/seedsigner #90: Validate passphrase

mandelbit, 2021-09-05:
It'd be useful to have a dedicated 'validate passphrase' option in the `Passphrase Action` menu, to double check a seed passphrase was set as intended after it's initially set.

mandelbit, 2021-09-05:
Alternatively/additionally, it could be useful to have a 'show passphrase' option in the `Seed Phrase` views (after selecting a seed in the temp storage)

kdmukai, 2021-09-06:
We're taking a super-close look at how we want to handle passphrases in our UI redesign. The basic principles for this new world are that you enter your passphrase when the seed is loaded or created, verify that it was entered right (w/a chance to edit if not), and then it's locked in. Can't display it again. Can't edit it. Can't add it later if you passed on the initial chance.

From that point on the seed would be displayed with an icon to indicate that it has a passphrase on it.

I'm also pushing to surface the master fingerprint more. That's the real verification that you'd want to make sure you entered your passphrase correctly (typo in passphrase = different/wrong fingerprint).

EverydayBitcoiner, 2022-12-27:
In my opinion, the only implementation of something like this that would make sense is to have a flow starting on the SeedOptionsView where you can reenter your passphrase and seedsigner will tell you whether it is correct or not. Though this would be a little bit at odds with the fact that the interface does not really indicate whether a seed was generated with a passprase or not (and it seems that the consensus is leaning towards keeping it that way https://github.com/SeedSigner/seedsigner/pull/286#issuecomment-1363363860).

I am wondering if when a user goes through the backup flow for a seed that was generated with a passphrase if there should be a prompt at the end for the user to verify their passphrase (similar to the step for verifying the seed words). Currently when you backup a seed that has a passphrase the only thing backed up are the seed words and there is no indication whether the seed was generated using a passphrase.

jwander14, 2023-01-26:
Related to this, it would also be nice to have an "edit seed phrase" as an option, but even after a correct phrase has been entered/approved. 

Use case: you have multiple wallets with different seed phrases from the original creation of the root fingerprint. You can import the original seed then add and/or change the phrases to access unique seeds/wallets. 

Thanks for what you do!

kdmukai, 2023-01-26:
> Use case: you have multiple wallets with different seed phrases from the original creation of the root fingerprint. You can import the original seed then add and/or change the phrases to access unique seeds/wallets.

In that case (well, in any case), I'd create a SeedQR so that the one seed can be quickly reloaded. At that point you can just keep scanning it back in and set a different BIP-39 passphrase each time.