SeedSigner/seedsigner #105: [ FEATURE ] Automated build script to easily make SeedSigner SD cards.

sys-shock, 2021-10-02:
Hello, I would like to start off by saying that I love the SeedSigner project and wanted to contribute to the project by writing a shell script that helps users make their own SD cards easily. To be short, I have run into issues in writing it as I am still new to writing code. As such I wanted to reach out here and see if anyone was interested in this as a concept and wanted to help make it work. The code is on github in a private repo as of right now and I would rather it stay private until I can ensure that it at least works once. If you are interested in helping let me know and I will share the repo with you. Thank you for your time and have a good day.

openoms, 2021-10-03:
This is a good idea, but don't be shy to show the code before it is ready. The public conversation does help a lot.
In a (draft) PR changes can be suggested and tested easily.
Would be happy to cast an eye on it.

See some (more complicated) examples:
https://github.com/openoms/joininbox/blob/master/build_joininbox.sh
https://github.com/rootzoll/raspiblitz/blob/dev/build_sdcard.sh

sys-shock, 2021-10-03:
Thank you for the response and for the offer of help. I have made the repo public and I have parsed the Raspiblitz build_sdcard.sh to find examples when I started but I got writing and stopped referring to it (guess I was being stubborn and wanted to get it to work myself). You are right, I should go back and read through these and see what is being done (why reinvent the wheel, right?).

The repo can be found here:

https://github.com/sys-shock/seedsigner_build

ajmotto, 2021-10-04:
Hey I will also take a look

sys-shock, 2021-10-05:
Awesome, thank you very much!

openoms, 2021-10-07:
Added guidance to verify the base image to start with: https://github.com/sys-shock/seedsigner_build/pull/1/files

Ideally the process should be done in  a clean environment like a fresh Ubuntu live. See an example of the building steps here: 
https://github.com/openoms/joininbox/blob/master/FAQ.md#build-the-sdcard-image

Yvo8YcQDQJkhQ, 2021-11-10:
Have you considered using a tool like buildroot, yocto or nixos? They should also allow for other cool features such as tiny image size, fast booting, boot into ramdisk (so that the sd card can be removed) and reproducible builds. 
I think buildroot is the easiest to start with and nixos might be the most advanced in regards of reproducible builds.
For buildroot with a raspi zero this might be a good place to start: https://armphibian.wordpress.com/2019/10/01/how-to-build-raspberry-pi-zero-w-buildroot-image/

SeedSigner, 2021-11-18:
We have someone working on buildroot, but the process has turned out to be more complex than we expected. I think a build script would be a great place to start.

DesobedienteTecnologico, 2022-06-09:
Probably now is possible easily:

**SeedSigner OS**:
https://github.com/SeedSigner/seedsigner-os

closed 2023-08-26 by newtonick