SeedSigner/seedsigner #115: Swap (memory-to-disk spillover) should not be attempted

alevchuk, 2021-11-03:
SeedSigner tried to load swap on startup.

1. It fails, but this adds several seconds of delay to startup.
2. If it were to succeed (I don’t know the reason for failure), then this violates the promise that private key never touches SD card

Proposal:
1. Uninstall (`apt remove`) dphys-swapfile package
2. Remove swap from /etc/fstab 

Extra: loading services like “Sound Card” should also not be attempted to reduce attack surface and speed up startup time

[image]

SeedSigner, 2021-11-08:
Appreciate your thoughts on this. As you accurately noted, we disable swap to help ensure that the contents of physical memory are not cached on the memory card, which could result in inadvertent recording of PKs/seeds there. The way you suggest to disable swap could be a better way. Am always also curious about little things we can do to cut some time from the boot process.

alevchuk, 2021-11-26:
I did some timing measurements and found that `apt remove `apt remove dphys-swapfile` does not seed up boot. It remove the "[FAILED]" message but boot is still ~50 seconds.
[image]

I realized I actually like the "[FAILED]" because it's the only one and an easy way to see that swap is not enabled.

# Why Closing?
So just closing, and i'll open a new issue for disabling other stuff like sound and wi-fi to try to speed up boot time.

# unrelated but possibly useful note about the Keyboard
i found that I cannot type "|" in the bash prompt, it makes it impossible to do things like `apt list | grep swap`.

it's because I have U.S. keyboard (which you probably do too) and Raspi comes configured for British keyboards by default. To fix this inconvinices, you can do this:

Edit /etc/default/keyboard (When attaching with Monitor and a US Keyboard, you may find that your not able to type things like "|". This is not a problem when going over SSH.)

Replace

```
XKBMODEL="pc105"
XKBLAYOUT="gb"
```

with

```
XKBMODEL="pc104"
XKBLAYOUT="us"
```

closed 2021-11-26 by alevchuk