SeedSigner/seedsigner #116: [Request] Public image to be shipped with WiFi and Bluetooth disabled

openoms, 2021-11-05:
Struggling to find any Raspberry Pi Zero v1.3 models so will need to get to build SeedSigners from the RPi Zero 2 W which has WiFi and bluetooth connectivity.

Having it disabled in the image by default would be a safer choice than either:
- produce an alternative image (used by fewer people)
- make custom modifications on the image from this repo
- make custom hardware modifications
- use RF shielding with metal / Faraday case
- leave WiFi and Bluetooth active 

Developers can reactivate manually or build their own image (common practice already).

Steps to disable WiFi and Bluetooth (can be tested and  added to the readme):

Run:
```
sudo systemctl disable wpa_supplicant.service
sudo ifconfig wlan0 down
```
and make sure: 
`dtoverlay=disable-bt`
is set in the `/boot/config.txt`

both will stay off after a restart.

as mentioned in: https://twitter.com/openoms/status/1454182385747124230?s=19

SeedSigner, 2021-11-09:
Agreed, it is time, especially as the project has grown and its become harder to source Zero 1.3's. Will be incorporating these steps into the manual process documentation, as well was the pre-assembled SeedSigner image. Thank you for detailing this here.

newtonick, 2021-11-18:
I recommend adding dtoverlay=pi3-disable-wifi to the config.txt as well

https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README

openoms, 2021-11-27:
Solved in the 0.4.5 release https://github.com/SeedSigner/seedsigner/releases/tag/0.4.5.
Thank you! Closing.

closed 2021-11-27 by openoms