SeedSigner/seedsigner #121: *Feature Request* Add simple snake game upon initial power on

MaxSikorski, 2021-11-17:
*This is a feature request*

How about adding the old but simple snake game upon turning on the device? Can play snake before entering SeedSigner main menu. 

This can be a deterrent to show case this device is a "home-made" video gaming device. Similar ArcBTC's Bowser wallet (link below to repo)

Bowser wallet: [](https://github.com/arcbtc/bowser-bitcoin-hardware-wallet)

Thoughts? I'll be adding more details here in due time.

RealDudePerson, 2022-01-27:
This is a decent idea. There should be some passcode selection to enter the main Seedsigner loop. For instance entering a specific name on the 'high score menu' will get you to the seed signer menu.

SeedSigner, 2022-01-28:
This is something that is still being considered after the push to get 0.5.0 out. It will likely be incarnated as some kind of soduku game, as we can reuse elements of the zooming mechanism that Keith made for the SeedQR transcription process. Stay tuned.