SeedSigner/seedsigner #126: [FEATURE] Create public addresses from seed words

ozomer, 2021-11-24:
Creating an xpub is great, but you need a computer/smartphone to calculate the addresses.
It could be useful to create the addresses directly on the SeedSigner (by choosing a seed and an address index).

For example:
- You want to create a new wallet quickly and load funds to it.
- You are worried that your computer/smarphone is generating fake addresses, hoping that you will load funds to them.
- You want to have an extremely private cold-storage with multiple addresses where the xpub was never loaded to any a device that was connected to the internet.

P.S. Creating public addresses from an xpub could be nice too, but please keep both actions a single-step process - I would want to create public addresses directly from the seed words, without creating the xpub, saving the xpub, and only then create the addresses.

mcroad, 2021-11-28:
Creating receive addresses on-device puts the responsibility to keep track of used or unused addresses on the user. User should not reuse addresses. SeedSigner does not have access to the chain and does not know which addresses have already been used. For this reason, generating addresses on-device is a potential foot-gun.

A better feature would be to validate receive addresses. Device scans an address qr and says whether it belongs to a selected xpriv.

SeedSigner, 2021-11-29:
Not sure if the potential danger of address re-use should prevent us from providing the feature. Also, we already have receive address validation, just scan an address from the main menu.

ozomer, 2021-12-01:
> Not sure if the potential danger of address re-use should prevent us from providing the feature. Also, we already have receive address validation, just scan an address from the main menu.

Nice, I didn't know that.

You could show a warning message about address-reusage privacy dangers - and also about using a high-index address without using many of the previous address (wallets will stop searching for balance in addresses when a series  of unused addresses).

ozomer, 2021-12-04:
> Not sure if the potential danger of address re-use should prevent us from providing the feature. Also, we already have receive address validation, just scan an address from the main menu.

How does this actually works without knowing the derivation path? What if this is address number 1000?

SeedSigner, 2021-12-04:
The current implementation just begins attempting to validate addresses (using the standard derivation paths for change/receive) from the zero index position, am hoping that in future implementations we can provide users with the ability to jump to a specific address index number to validate.

kdmukai, 2022-05-01:
v0.5.0 includes a "Skip 10" feature.

I'd still like to see an Address Explorer feature added. Something like:

From Seeds menu:
* Select the seed you want to work with.
* Select "Address Explorer"
* Choices: Single Sig, Multisig
* Single Sig:
    * Select script type (w/Custom Derivation as an option)
* Multisig:
    * Prompt to load the multisig wallet descriptor via QR
* Prompt for "Receive" or "Change" addr
* Show address reuse warning
* Maybe options: "Browse list", "Jump to index"
* Browse list:
    * Shows next 5 addrs as a button list (clicking on button displays addr details).
    * Bottom button is "Next"
    * If the "Next" page's first index is greater than typical wallet gap (e.g. 20?), show a wallet gap warning.
* Jump to index:
    * Leads to Keyboard UI to type in the index
    * If the index is greater than typical wallet gap (e.g. 20?), show a wallet gap warning.
    * Shows the Browse list w/first result at the specified index. 
* Address details screen:
    * Title: `Receive #15` or `Change #23`
    * Show QR code
    * Show address as a `FormattedAddress` (highlights first/last 7 chars)

EverydayBitcoiner, 2022-12-27:
Still to do from above list:
- Custom Derivation option (This seems like an pretty advanced power user feature, not sure how much demand there is for this)
- Address reuse warning
- Options: "Browse List",  "jump to index"
- Address details Screen (currently jumps straight to qr export)

Are these remaining points being worked on by anyone?

It looks like everything else from the list is working in the dev branch.

Also issue https://github.com/SeedSigner/seedsigner/issues/216 was submitted for getting receive addresses using xpub or singlesig wallet descriptor. Wondering if that issue should be moved into this one?