SeedSigner/seedsigner #127: Disable wi-fi and other services to speed up boot time

alevchuk, 2021-11-26:
Wi-fi is not needed (because we purposely use hardware without wifi) but loading the wi-fi service is potentially a slowdown of the boot process.

btw, Wi-Fi shows up as "WPA Supplicant"

Might as well disable other services that are not needed, such as mDNS and Sound card stuff:
[image]

SeedSigner, 2021-11-29:
Will note this for the next release, appreciate it.

alevchuk, 2022-01-04:
I made an additional  1.3x speed up in startup time: https://link.medium.com/x36uxWE4xmb

SeedSigner, 2022-01-16:
Going to look at these additional boot time reductions for the next release, thank you for putting them out there.

Marc-Gee, 2023-02-28:
Check out the speed of startup on a Pi Zero 2 using version 0.6.0, on the new [no wifi] minimized Seedsigner_OS ! 
14 seconds startup time for me! :))

closed 2023-02-28 by newtonick