SeedSigner/seedsigner #128: [FEATURE] settings.ini from boot (FAT32) partition 📝

DesobedienteTecnologico, 2021-12-14:
[image]

### To make that possible you have three ways right now:

1. 🩹 Apply a [Git Patch](https://pastebin.com/M4jPTCQ3)
2. 📥 Clone my [Fork](https://github.com/DesobedienteTecnologico/seedsigner)  
3. ✂️ Copy/Paste from the [Fork](https://github.com/DesobedienteTecnologico/seedsigner), the directory **seedsigner/src/**  once you all downloaded in your computer.

My fork is here: https://github.com/DesobedienteTecnologico/seedsigner

#### If you would like to have this function in future versions (Before custom OS get released). Please open a new branch to push changes :) .  

---
## 🪛 After you have applied changes commented before:

We need to make users able to read/write files from /boot:

1. Go **/etc/** in **rootfs** MicroSD partition.
2. Open with root permisions **fstab**.
3. Add **,umask=000** after defaults as I show in the picture.
4. Save changes :) .

[image]

Put in the **boot** MicroSD partition the file **settings.ini** and ENJOY! :) 

---
## 🧐 Details:

- **seetings.ini** is now working from the **boot** MicroSD partition.
- Due the boot partition is **FAT32**... Windows, MacOS, Android and Linux can work with it easily.
- When system boots, the first partition called **boot** is mounted automatically from **/dev/mmcblkp1** to **/boot** in the system.

---

## Info:

[How to apply a Git patch](https://www.git-tower.com/learn/git/faq/create-and-apply-patch/)

newtonick, 2021-12-14:
Looks like the goal of this feature is to move the settings.ini to the boot partition to make it easier for seedsigner users to modify directly. I think this is a good idea for experimenters and tinkerers. I like the idea to moving this file to the boot partition and think it would be worth pursuing in the next seedsigner release.

I don't think it's a good idea to advise seedsigner users to applying git patches and pulling code from outside the official repo. This increases risk and can cause confusion.

I didn't see a PR for this, @DesobedienteTecnologico did you plan on creating one? I'd be happy to work on this if not.

DesobedienteTecnologico, 2021-12-14:
> I didn't see a PR for this, @DesobedienteTecnologico did you plan on creating one? I'd be happy to work on this if not.

Yes sure. I just did the PR https://github.com/SeedSigner/seedsigner/pull/129

newtonick, 2023-08-26:
Working in SeedSigner OS

closed 2023-08-26 by newtonick