SeedSigner/seedsigner #130: QR will not scan

jgoodz, 2021-12-16:
Recently bought all the hardware for 2 SeedSigners on my own. Upon setting it up I am unable to scan QR codes with the camera. It turns on and I can see the QR code on screen but it deosnt detected that its a QR code. I've tried Low Med and High camera density settings. Image seems a little blurry. I tried small close up QR, larger further back QR, no dice.

Using Raspberry Pi Zero (no wifi)
Using recommended LCD Hat

Camera: Raspberry Pi Camera Rev1.3 P5V04A SUNNY

Does anyone have an idea who to get it to scan properly? This camera seems pretty common. Has anyone else had luck with this model?

Thanks

SeedSigner, 2021-12-16:
It sounds like a focus issue with the camera. The great majority of cameras come out of the box focused well enough that they scan fine, but occasionally they show up out of focus. Some in our Telegram community have had success with this technique, though it is possible to damage the lens:

https://www.jeffgeerling.com/blog/2017/fixing-blurry-focus-on-some-raspberry-pi-camera-v2-models

Are both of your builds experiencing the same issue?

jgoodz, 2021-12-16:
I've been trying to adjust the focus with no luck I need a tool to adjust it. I see 3d printed tools online but they seem to be for a different camera as mine has 3 divets to grip around while the tools match 4 divet pattern. I will check out this link you provided. 

In the attached pic it seems like it's clear enough but no detection. I tried to press all buttons with no success either. I put random qr code on a 43 inch tv and was about 4 or 5 feet away. Maybe the focus at further distances trigger the qr recognition?

Thanks. 

[image]

SeedSigner, 2021-12-16:
Can you say what kind of QR code that is that you're scanning? What software generated it? Also, does QR scanning operate ok in the test module?

jgoodz, 2021-12-16:
My apologies, I think I realize my problem. My assumption was that this would also be able to scan a private key qr code from a wallet generated on the fly from bitaddress.org and segwitaddress.org, as this would be similar to my intended use case off a paper wallet. I did not realize writing the QR on paper after generating by the seedsigner was the only option.

SeedSigner, 2022-01-16:
Closing this, thank you for following up!

closed 2022-01-16 by SeedSigner