SeedSigner/seedsigner #131: [FEATURE] BIP85 Support

kornpow, 2021-12-22:
I thought it would be really cool if SeedSigner could support BIP85, which is a method for generating a derived set of seed words/ entropy from a starting seed words.

Would be great if you could:
1. Scan in a SeedQR, save into one of the storage slots
2. Derive BIP85 seeds/entropy, and either allow export or save into other seed slots.

This BIP is currently only implemented in ColdCard at the moment. I've used BIP85 entropy in a few different ways so far, which unlock some cool use cases:
1. Export an `xprv...` private key, and use for initializing LND, instead of the AEZEED cipher
2. Export seed words, and use those seed words for software wallet (phone wallets, or Sparrow wallet for Whirlpool)
3. Create a multisig wallet, using only a single seed ie. making a multisig with derived seed index 0, 5, 10. Basically a multisig hidden in a single sig

Marcsmyname, 2022-02-13:
I like this idea. I've recently started going down this BIP85 rabbit hole. With 10,000 possibilities, you can even use one key as a passphrase for your wifi. (you obviously wouldn't send any bitcoin to addresses derived from this xpub) 

I'd like to have this feature just because it makes it easier to verify the cold card.

I wish I would have thought of using one for LND.

jase100k, 2022-03-26:
BIP85 + passphrase support would really be great here.  

I hate having to right down new seeds all the time and securing them, and trying to stamp seeds on to metal each time.  Can also be very expensive if you don't use your own materials sourced like washers.

Another use case I can see is that you can protect the seed, and then derive new seeds for your kjds or parents, all from one seed.   

One seed to rule them all.

Semisol, 2022-04-01:
Would be really useful to have this, to for example have one securely stored seed to generate multiple hot wallets.

SeedSigner, 2022-04-04:
We appreciate the comments in here -- people are definitely passionate about BIP85.  : )   We would also need to se if embit supports BIP85 since we use it for most of our interaction with the protocol...

jase100k, 2022-04-07:
Here's a simple BIP85 implementation in seedsigner developed on branch 0.5.0-pre3.  Only main issue is it requires BIP85 library https://github.com/ethankosakovsky/bip85

i.e pip install bip85

source code at https://github.com/jase100k/seedsigner/tree/0.5.0-pre3

1. Scan your SeedQR
2. Add your passphrase if necessary
3. There's an option in the seed page to Gen Bip85 Seed select that
4. Choose 12/24 words (18 words does anyone use?)
5. Choose your index 
6. Warning message displays around seeds
7. BIP85 child seed is displayed ( 4 words per page)

The other 3 methods to derive entropy, i.e xprv, wif, seed are not done yet.  I don't like how the index entered needs to be remembered by the user.  Use test seeds.

Semisol, 2022-04-08:
> We appreciate the comments in here -- people are definitely passionate about BIP85. : ) We would also need to se if embit supports BIP85 since we use it for most of our interaction with the protocol...

nope as far as I'm aware, but you can see a bip85 implementation above.

SeedSigner, 2022-04-12:
Just want to flag @kdmukai and/or @newtonick on this to see if they're willing to review the implementation and share any perspective. Appreciate your work on this.

Semisol, 2022-04-20:
Any status updates?

kdmukai, 2022-04-21:
@jase100k closed PR #172 but didn't leave a comment.

That PR added an external pip dependency, but it was written by the BIP-85 author as a reference implementation so seems reasonable to include.

If the PR is revived against the final v0.5.0 release, I'd be happy to dive into it.

jase100k, 2022-04-23:
PR #194 updated to remove bip85 reference library dependency.

EverydayBitcoiner, 2022-12-29:
The mentioned PR #194 was merged three weeks ago into the dev branch.

jdlcdl, 2022-12-29:
@EverydayBitcoiner, I'm really enjoying watching how you've hit the ground running here, with code, with comments, with links to other issues/prs, all, im assuming, with intention to get another task all the way to "Done!".

Nice work!  Way to close out the year!  All the best in the next one.

closed 2023-02-28 by newtonick