SeedSigner/seedsigner #132: [Feature] Option to add last bits of entropy to "Calculate Final Word"

kdmukai, 2021-12-23:
Per BIP-39, a 12-word seed the final word includes 7 bits of entropy plus the 4-bit checksum. But when picking 11 words out of a hat, there's no way to specify those last 7 bits. The final word calculation just defaults those 7 bits to all zeroes. So your resulting entropy is 121 bits instead of 128.

In a 24-word seed the final word includes 3 bits of entropy plus 8-bit checksum.

Proposal: Add an "Advanced" option that prompts for 7 or 3 coin flips to complete the last remaining bits of entropy.

Realistically most users won't care about this (nor understand it--I certainly wouldn't have until recently). But cool to have as an option?

aitor3ml, 2022-01-15:
Could we input the last word (number 12 or 24) and then let the seedsigner "correct" it by taking those bits, calculating the checksum and giving back the last word?
We would be inserting those 7 or 3 bits choosing one of the words starting with the chosen bits.

SeedSigner, 2022-01-16:
I would imagine there should be some kind of workaround with embit as it currently exists to be able to add the additional inputs. Might need to jump through some hoops in terms of converting the PK to a different format / number base to accept the supplemental inputs?

aitor3ml, 2022-01-16:
> I would imagine there should be some kind of workaround with embit as it currently exists to be able to add the additional inputs. Might need to jump through some hoops in terms of converting the PK to a different format / number base to accept the supplemental inputs?

As I tried to explain before, I would feel comfortable using any word with the desired bits and letting the seedsigner "correct" the checksum and giving back the right word. I would still be able to check that the given word begins with the chosen bits.

Since my purpose is to calculate the checksum for an offline created PK this fulfills my requirements, but I don't know if others are aiming for other goals or you can think in a more suitable way to enter the PK without the checksum.

EverydayBitcoiner, 2022-12-30:
PR #197 implements the discussed changes.

kdmukai, 2023-01-22:
Closed by #197.

closed 2023-01-22 by kdmukai