SeedSigner/seedsigner #133: Invalid 3D enclosure files

BenGWeeks, 2021-12-28:
Both PrusaSlicer and Infusion 360 have errors when trying to open the STL and F3D files respectively.

PrusaSlicer gives (for all *.stl files):

```
PrusaSlicer has encountered an error
Loading of a model file failed.
```

And Infusion 360 gives (for all *.f3d files):

```
The path C:/Users/*****/AppData/Local/Autodesk/Autodesk Fusion 360/*****/W.login/F/_OP_Lower.9fae7673-a594-4304-a112-c846235622a4.f3d is not a valid zip package.
```

SeedSigner, 2021-12-28:
I was just able to download the Open Pill .stl files and they loaded into Prusa Slicer just fine. I have also spoken with people who have been able to successfully download and work with the Fusion 360 files. Perhaps try downloading the entire repo as a .zip file and extracting the files you're seeking that way?

BenGWeeks, 2021-12-28:
My bad, solved by not doing "save link as" (which still prompted to save as STL) but clicking on the STL file then clicking the "Download" button in GitHub. Thanks for your swift assistance!

SeedSigner, 2021-12-28:
Glad it's resolved!

closed 2021-12-28 by SeedSigner