SeedSigner/seedsigner #134: Seed generated with dice rolls on 0.4.5 does not match seed generated with the coldcard rolls.py script

eriknylund, 2022-01-04:
With the introduction of https://github.com/SeedSigner/seedsigner/pull/109 I am following the guide on https://coldcard.com/docs/verifying-dice-roll-math to verify that the SeedSigner 0.4.5 and the Coldcard produce the same seed and I am seeing some differences.

# Steps to reproduce

On SeedSigner 0.4.5 go to Seed Tools > Generate Seed with Dice

Enter 45 twos, followed by 54 fives. This will give you a seed `1. garden 2. uphold ... 23. feature 24. wave`. 

Perform the same using rolls.py and you will see that the seeds match:
```
% echo 222222222222222222222222222222222222222222222555555555555555555555555555555555555555555555555555555 | python3 rolls.py 
5fbdda0295cdc4c682fbb5eba6d704b90f281a45f6eab283e8b7c86abab4d50f

   1: garden
   2: uphold
   3: level
   4: clog
   5: sword
   6: globe
   7: armor
   8: issue
   9: two
  10: cute
  11: scorpion
  12: improve
  13: verb
  14: artwork
  15: blind
  16: tail
  17: raw
  18: butter
  19: combine
  20: move
  21: produce
  22: foil
  23: feature
  24: wave
```

Repeat the test above, but replace the last five with a six. The SS will display

`1. whale 2. monitor ... 23. capital 24. owner`

However, this time rolls.py differs:

```
% echo 222222222222222222222222222222222222222222222555555555555555555555555555555555555555555555555555556 | python3 rolls.py 
82c39211f153aea293a41d03ea77c619170d3f552d378dd69467f6c5eea58f12

   1: lizard
   2: broken
   3: love
   4: tired
   5: depend
   6: eyebrow
   7: excess
   8: lonely
   9: advance
  10: father
  11: various
  12: cram
  13: ignore
  14: panic
  15: feed
  16: plunge
  17: miss
  18: regret
  19: boring
  20: unique
  21: galaxy
  22: fan
  23: detail
  24: fly
```

# Expected output 

I would expect that the seed generated by SS 0.4.5 and rolls.py match for all dice rolls.

eriknylund, 2022-01-05:
I added tests for these and a few more in https://github.com/SeedSigner/seedsigner/commit/451757a0d93a5c4540b5d724015c15ecf151833b and the tests run without issues, which is great but also confusing 🤔 

Is someone able to reproduce the diff I'm seeing on the my device?

[image]

`222222222222222222222222222222222222222222222555555555555555555555555555555555555555555555555555556`

[image]

eriknylund, 2022-01-05:
Found the line that is causing issues here https://github.com/SeedSigner/seedsigner/blob/main/src/seedsigner/views/seed_tools_view.py#L798 and it can be verified through rolls.py by replacing the 6 with a 0:

```
% echo 222222222222222222222222222222222222222222222555555555555555555555555555555555555555555555555555550 | python3 rolls.py 
f9b1e195b5360528c59a8d2dabf24872bd05c15ebbf10f4cd233b3c94d168884

   1: whale
   2: monitor
   3: grace
   4: heavy
   5: gate
   6: nerve
   7: biology
   8: faculty
   9: color
  10: garbage
  11: must
  12: tornado
  13: space
  14: thing
  15: kit
  16: lawsuit
  17: aunt
  18: grief
  19: mind
  20: guide
  21: network
  22: mercy
  23: capital
  24: owner
```

So, a potential fix would be to remove L798-L800.

eriknylund, 2022-01-06:
PR #135 fixes the issue.

SeedSigner, 2022-01-16:
Leaving this open until the fix is in a release, thank you for your work on this issue.

UncleSamtoshi, 2022-01-27:
Am I correct that this doesn't effect the security of seeds generated with dice rolls but rather just creates a non standard output?

eriknylund, 2022-01-27:
> Am I correct that this doesn't effect the security of seeds generated with dice rolls but rather just creates a non standard output?

That is correct. The security is not impacted. If you have your dice rolls stored, you can still verify it using the rolls.py by replacing every 6 with 0 before passing it in, as can be see in the example a few comments above, where the final dice roll is 0.

UncleSamtoshi, 2022-01-27:
Awesome thanks!

SeedSigner, 2022-02-12:
Thank you for following up with this testing, we are going to have to make sure we have this module reliably matching what the other two methods come up with before the next release goes out.

eriknylund, 2022-02-13:
@sovereignhodler Thanks for taking your time with a more complex set of dice rolls ❤️ I tried to reproduce this on my end and I get the same output as the Seedsigner 0.4.5 (when replacing the 6 with 0):
```
% python3 --version
Python 3.9.8

% echo -n 305410545242515003010302453535501111032153013053535515044520322014054350021501403314304111401333132 | python3 rolls.py 
88af342fb10b439cac817f6636407c5367ed76b5c5fcbe38e8d478d9de8d99bc

   1: master
   2: keen
   3: armor
   4: ginger
   5: reflect
   6: soft
   7: rare
   8: copper
   9: great
  10: rare
  11: business
  12: plug
  13: learn
  14: robust
  15: strike
  16: sauce
  17: sail
  18: shrug
  19: crush
  20: vast
  21: over
  22: egg
  23: group
  24: measure
```

Could you post your Python version and maybe we can figure out whats different on your setup? On my machine Python 2 is the default, but I couldn't even get the script to run there, so that can't be it:

```
% python --version
Python 2.7.16

% echo -n 305410545242515003010302453535501111032153013053535515044520322014054350021501403314304111401333132 | python rolls.py 
Traceback (most recent call last):
  File "rolls.py", line 12, in <module>
    r = input().strip()
AttributeError: 'long' object has no attribute 'strip'
```

Also, to make sure we're running the same rolls.py, here's my sha256 for it:
```
% shasum -a 256 rolls.py                                           
01189cf6f54da328bafadf959ab5ec6ec3acd0ffbe47b042f2085e249ee660f4  rolls.py
```

sovereignhodler, 2022-02-16:
I ran the rolls.py script from Git Bash on my W10 machines and it gives me the same mnemonnic there. Assuming this is a local issue with my python installation or with the rolls.py script for now

SeedSigner, 2022-03-02:
Closing this out as it has been resolved in 0.4.6; here is a brief thread on cross-validating the dice-to-seed process:

https://twitter.com/SeedSigner/status/1496321570008145920?s=20&t=N3pMMMi4yfXxY4TMJNxjBg

closed 2022-03-02 by SeedSigner