SeedSigner/seedsigner #137: Manually enter 24 word seed not saved

jessjamesjones, 2022-01-16:
When I manually enter 24 word seed in "Temp Seed Storage", the seed is not saved, instead I get returned to "Temp Seed Storage" page

asvdf, 2022-01-27:
90% chance it's because the last word is not a valid checksum, are you sure you entered it 100% correct, and it is a BIP39 compliant seed? For example aezeed wouldn't work on SS

jessjamesjones, 2022-01-28:
Tested 5 times now with my seed generated from Umbrel. It dosent say that
the seed is invalid, just get returned to the menu.
Same happened with the seed from Blixt wallet.

asvdf, 2022-01-28:
Wouldn't know about Blixt, but Umbrel lightning node uses lnd, which in turn uses aezeed, which is not BIP39 compatible

jessjamesjones, 2022-01-28:
Now I understand, thanks for the info.

SeedSigner, 2022-01-28:
I have also recently encountered this issue when entering a seed that was generated with Electrum, which is also not BIP39 compatible. While we may not support alternate seed implementations any time soon, we can do better with the messaging when this issue arises, so I am going to leave this issue open as a reminder.

jessjamesjones, 2022-01-28:
Great, thanks for the info, great support and amazing product.

SeedSigner, 2022-02-12:
Thank you for looking into it on their end as well.

kdmukai, 2022-05-01:
v0.5.0 now has an "Invalid Mnemonic!" warning screen that lets you either go back and edit your seed phrase or discard it.

@newtonick you can close this issue.

jessjamesjones, 2022-05-02:
Done

closed 2022-05-02 by jessjamesjones