SeedSigner/seedsigner #141: Some feature ideas and a "bug" with passphrase display

asvdf, 2022-01-29:
My post in telegram group didn't receive any ACK, be it positive or negative, so probably this is the more appropriate place to write.

1) allow for "hold joystick pressed" to keep moving the cursor back and forth in passphrase entering UI (to rapidly check passphrase was entered correctly before saving)
2) have the cursor in passphrase entering UI blink, instead of fixed (catches the eye better to understand where it's at, and it's something users are familiar with on practically every OS)
3) display spaces in passphrase (both input and results UI) in a way they are discreet, maybe using a small dot, an empty box, something not among the allowed passphrase chars, that will show how many (if any) spaces are present especially at line wrap
5) during seed input UI, instead of resetting the cursor position to a at the beginning of the next word, leave it at the same position of the last letter entered (flows better while inserting, as you might be already thinking about the next word, and know what movements you need to do to get there, but when actually on next word the cursor will be moved on top left corner instead, breaking continuity; also, having it closer to the middle will statistically improve chances that the next letter is closer to reach than from any corner)

The "bug" I found: after entering passphrase for a seed saved in temp storage, said passphrase will be displayed wrapped at the edges of the screen so it's 100% visible (and BTW it would be useful to show spaces with some kind of symbol in here, so for spaces falling just at the edge you can distinguish them clearly); if entering seed+passphrase in response to scanning a QR (address to verify, or psbt to sign), with the seed storage empty, after having entered the passphrase, it will be shows all in one line, unwrapped, with beginning and ending outside of the screen, leaving the use unable to confirm it's been entered correctly.

SeedSigner, 2022-01-29:
Thank you for leaving these ideas here, and yes, the chat room is a more informal conversation where people can ask questions and engage in casual chat. Having these here is definitely helpful. We're in the midst of a big push to get 0.5.0 coded and into production, and as we start to refine things after the release, observations like these are super helpful in the way of user feedback.

jdlcdl, 2023-01-27:
> 3\. display spaces in passphrase (both input and results UI) in a way they are discreet, maybe using a small dot, an empty box, something not among the allowed passphrase chars, that will show how many (if any) spaces are present especially at line wrap

For strange spaces in the passphrase, as of merged changes on Jan 16th 2023, in issue: https://github.com/SeedSigner/seedsigner/pull/295,

If the passphrase contains leading-spaces OR trailing-spaces OR multiple-consecutive-spaces (also nothing but spaces), then all spaces in the passphrase will be replaced with left-7/8s-block characters (looks like a solid cursor but doesn't run into the next one when consecutive) during the passphrase review.

If the passphrase does not have these hard-to-see spaces, then it will appear unchanged and wrapped as it has in the past.

Great input @GordianLN, we still have more improvements to make on this open issue.