SeedSigner/seedsigner #142: Auto shutdown after N minutes

asvdf, 2022-01-30:
I find myself leaving the SS turned on on my desk with the usb cable connected.
Bad habit which I need to lose, yet I see how others might fall to the same.
A function to auto-turnoff after 20 minutes, maybe less, maybe more, could add to the security of the device.

jdlcdl, 2022-11-21:
I too like the idea of a user settings adjustment as a security solution to forgetting about the seedsigner being left on.
I'd like to hear what others prefer as to how to go about that solution... shutdown? or restart? or forget seeds? or demand a pin?

There may also be an issue of wasted energy/usage when idle (if on battery pack)?   I've been turning off the backlight for this personally, it keeps the rpi2 cooler and doesnt bump its cpu-usage from 2% to 60% like the screensaver does.  If this interests anyone, take a look at my repo's screensaver_blankscreen branch commit.

GordianLN, 2022-11-21:
Nice input.
There can be a safety+convenient feature (still better than leaving the SS on at the mercy of anyone) where operation is "locked", and you have the option to shutdown, continue operation purging seeds, or restore seed access by entering the Mth and Nth word, asked at random, so you at least save yourself the pain of entering the whole seed again in case of automatic shitdown

BtcAutoNode, 2024-11-27:
Hm, if the seedsigner is left active on the desk, how far away is the seed or qr which had been used before to be loaded into seedsigner. 
The time of the timer needs to be elapsed until the auto-lock function works...what happens within this time-frame. Or you need to do an action like lock-screen but then you could also have been pulling the power cable before leaving.

I am thinking about a pin s.o. sets on the device (once at first usage or every time sesi is switched on) which will be requested for all actions where the seed is involved (like exporting the seed words or qr / signing a tx / ...). You can go to the settings and change the pin but then need the old one or if you forget it you need to re-flash the software and 1st start will ask you again to set a pin (or any action involving the seed can tell you that there is no pin yet). Something like that just to make sure that someone can not easily get access to such sensitive information if the door bell rings and you go to the door but have visitors (or other people living there) at home.
If the sesi is powered with a power bank (portable power device), then someone could take it with him while powered on and try all the possible combinations, so the pin number must be a bigger one, 6-7 digits, idk.