SeedSigner/seedsigner #144: [FEATURE] Use SeedSigner to sign messages

RobDoesData, 2022-02-14:
With SeedSigner as an economical air gapped signing device, it would be interesting to enable the ability to not only sign PSBTs, but user generated messages. 

Proposal: Add under an advanced menu an option to sign a message once.

User Flow:

A user would either Load Seed QR Code or QR generation or text message in QR form:

Text Message Format (two lines):
Line 1: Message to be signed
Line 2: Derivation path of what key is signing that message.

This is a similar approach to how cold card handles signing of messages: (https://coldcard.com/docs/microsd#sign-text-file)

With message payload and seed phrase loaded, it would provide a prompt of:
1. Message Content
2. Key that is signing

Once accepted, the seedsigner would sign the transaction and present the signed message on screen.

SeedSigner, 2022-02-14:
This feature has been discussed by myself and Nick & Keith, and I believe we are on board, just need to work it into our queue of planned features.

jdlcdl, 2022-11-21:
It is my belief that https://github.com/SeedSigner/seedsigner/pull/262 is meant to address this issue.

bitcoinprecept, 2023-05-09:
This topic is also mentioned in https://github.com/SeedSigner/seedsigner/issues/254

closed 2024-08-11 by newtonick