SeedSigner/seedsigner #149: bullseye build missing dependencies

nyxnor, 2022-02-19:
I know the readme clearly states to use the buster image, but buster is the old stable, bullseye is the stable now. Old software that will be left with unfixed bugs.

**wiringpi**
- deprecated
 - Is any of [these alternatives](https://raspberrypi.stackexchange.com/a/133252) viable to substitute wiringpi?
- raspi-gpio was already installed so now I am trying to use `libgpiod2` to see if it works

**python-pil**
- python2 deprecated
- testing with `python3-pil` on debian 11.

```
pi@raspberrypi:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
```
pi@raspberrypi:~ $ sudo apt-get update -y && sudo apt-get install -y wiringpi python3-pip python3-numpy python-pil libopenjp2-7 git python3-opencv libzbar0 python3-picamera libatlas-base-dev qrencode
Get:1 http://archive.raspberrypi.org/debian bullseye InRelease [23.6 kB]                                        
Get:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB]                                     
Get:3 http://archive.raspberrypi.org/debian bullseye/main armhf Packages [251 kB]
Get:4 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf Packages [13.2 MB]
Fetched 13.5 MB in 37s (361 kB/s)                                                                                                     
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package wiringpi is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package python-pil is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'wiringpi' has no installation candidate
E: Package 'python-pil' has no installation candidate
```

SeedSigner, 2022-02-19:
The Pi Foundation broke a lot of things that we use with bullseye. I completely understand your point; as bullseye matures and they fix some of the things they broke, we will re-evaluate moving back to their more current releases.

nyxnor, 2022-02-19:
Just opened the issue because keywords `bullseye` and `debian 11` did not appear and hopefully get bullseye to work.

nyxnor, 2022-02-19:
```sh
Building wheels for collected packages: RPi.GPIO
  Building wheel for RPi.GPIO (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [86 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-armv7l-3.9
      creating build/lib.linux-armv7l-3.9/RPi
      copying RPi/__init__.py -> build/lib.linux-armv7l-3.9/RPi
      creating build/lib.linux-armv7l-3.9/RPi/GPIO
      copying RPi/GPIO/__init__.py -> build/lib.linux-armv7l-3.9/RPi/GPIO
      running build_ext
      creating build/temp.linux-armv7l-3.9
      creating build/temp.linux-armv7l-3.9/source
      arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/home/pi/.envs/seedsigner-env/include -I/usr/include/python3.9 -c source/c_gpio.c -o build/temp.linux-armv7l-3.9/source/c_gpio.o
      arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/home/pi/.envs/seedsigner-env/include -I/usr/include/python3.9 -c source/common.c -o build/temp.linux-armv7l-3.9/source/common.o
      arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/home/pi/.envs/seedsigner-env/include -I/usr/include/python3.9 -c source/constants.c -o build/temp.linux-armv7l-3.9/source/constants.o
      arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/home/pi/.envs/seedsigner-env/include -I/usr/include/python3.9 -c source/cpuinfo.c -o build/temp.linux-armv7l-3.9/source/cpuinfo.o
      arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/home/pi/.envs/seedsigner-env/include -I/usr/include/python3.9 -c source/event_gpio.c -o build/temp.linux-armv7l-3.9/source/event_gpio.o
      arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/home/pi/.envs/seedsigner-env/include -I/usr/include/python3.9 -c source/py_gpio.c -o build/temp.linux-armv7l-3.9/source/py_gpio.o
      source/py_gpio.c: In function ‘PyInit__GPIO’:
      source/py_gpio.c:1046:4: warning: ‘PyEval_ThreadsInitialized’ is deprecated [-Wdeprecated-declarations]
       1046 |    if (!PyEval_ThreadsInitialized())
            |    ^~
      In file included from /usr/include/python3.9/Python.h:145,
                       from source/py_gpio.c:23:
      /usr/include/python3.9/ceval.h:129:36: note: declared here
        129 | Py_DEPRECATED(3.9) PyAPI_FUNC(int) PyEval_ThreadsInitialized(void);
            |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
      source/py_gpio.c:1047:7: warning: ‘PyEval_InitThreads’ is deprecated [-Wdeprecated-declarations]
       1047 |       PyEval_InitThreads();
            |       ^~~~~~~~~~~~~~~~~~
      In file included from /usr/include/python3.9/Python.h:145,
                       from source/py_gpio.c:23:
      /usr/include/python3.9/ceval.h:130:37: note: declared here
        130 | Py_DEPRECATED(3.9) PyAPI_FUNC(void) PyEval_InitThreads(void);
            |                                     ^~~~~~~~~~~~~~~~~~
      arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/home/pi/.envs/seedsigner-env/include -I/usr/include/python3.9 -c source/py_pwm.c -o build/temp.linux-armv7l-3.9/source/py_pwm.o
      arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/home/pi/.envs/seedsigner-env/include -I/usr/include/python3.9 -c source/soft_pwm.c -o build/temp.linux-armv7l-3.9/source/soft_pwm.o
      arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 build/temp.linux-armv7l-3.9/source/c_gpio.o build/temp.linux-armv7l-3.9/source/common.o build/temp.linux-armv7l-3.9/source/constants.o build/temp.linux-armv7l-3.9/source/cpuinfo.o build/temp.linux-armv7l-3.9/source/event_gpio.o build/temp.linux-armv7l-3.9/source/py_gpio.o build/temp.linux-armv7l-3.9/source/py_pwm.o build/temp.linux-armv7l-3.9/source/soft_pwm.o -L/usr/lib -o build/lib.linux-armv7l-3.9/RPi/_GPIO.cpython-39-arm-linux-gnueabihf.so
      /usr/bin/ld: build/temp.linux-armv7l-3.9/source/constants.o:/tmp/pip-install-kbwdl8nz/rpi-gpio_58a6e4a3d5c04cd993f77f2f7144e4c7/source/common.h:41: multiple definition of `module_setup'; build/temp.linux-armv7l-3.9/source/common.o:/tmp/pip-install-kbwdl8nz/rpi-gpio_58a6e4a3d5c04cd9 [... cut ...]
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 66.3/66.3 KB 95.3 kB/s eta 0:00:00
Collecting six==1.16.0
  Using cached https://www.piwheels.org/simple/six/six-1.16.0-py2.py3-none-any.whl (11 kB)
Requirement already satisfied: spidev==3.5 in /home/pi/.envs/seedsigner-env/lib/python3.9/site-packages (from -r requirements.txt (line 9)) (3.5)
Installing collected packages: RPi.GPIO, pyzbar, picamera, embit, six, Pillow, numpy, qrcode
Successfully installed Pillow-8.2.0 RPi.GPIO-0.7.1 embit-0.4.10 numpy-1.21.1 picamera-1.13 pyzbar-0.1.8 qrcode-6.1 six-1.16.0
```

nyxnor, 2022-02-19:
Tests failing.
```sh
(seedsigner-env) pi@raspberrypi:~/seedsigner $ pytest
========================================================= test session starts =========================================================
platform linux -- Python 3.9.2, pytest-6.2.4, py-1.11.0, pluggy-0.13.1
rootdir: /home/pi/seedsigner, configfile: pytest.ini, testpaths: tests
collected 14 items / 1 error / 13 selected                                                                                            

=============================================================== ERRORS ================================================================
_____________________________________________ ERROR collecting tests/test_decodepsbtqr.py _____________________________________________
../.envs/seedsigner-env/lib/python3.9/site-packages/_pytest/python.py:578: in _importtestmodule
    mod = import_path(self.fspath, mode=importmode)
../.envs/seedsigner-env/lib/python3.9/site-packages/_pytest/pathlib.py:524: in import_path
    importlib.import_module(module_name)
/usr/lib/python3.9/importlib/__init__.py:127: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
<frozen importlib._bootstrap>:1030: in _gcd_import
    ???
<frozen importlib._bootstrap>:1007: in _find_and_load
    ???
<frozen importlib._bootstrap>:986: in _find_and_load_unlocked
    ???
<frozen importlib._bootstrap>:680: in _load_unlocked
    ???
../.envs/seedsigner-env/lib/python3.9/site-packages/_pytest/assertion/rewrite.py:161: in exec_module
    source_stat, co = _rewrite_test(fn, self.config)
../.envs/seedsigner-env/lib/python3.9/site-packages/_pytest/assertion/rewrite.py:354: in _rewrite_test
    tree = ast.parse(source, filename=fn_)
/usr/lib/python3.9/ast.py:50: in parse
    return compile(source, filename, mode, flags,
E     File "/home/pi/seedsigner/tests/test_decodepsbtqr.py", line 435
E       1_part = '{"label": "Testnet Single Zone", "blockheight": 2090512, "descriptor": "wpkh([990a73ad/84h/1h/0h]tpubDDHQMDnFdan2GyHBsG32VW9qiygbhVizGRTjiS3H79M49FSvpsvLXqLgp1yC7r43dXVHozWavi2Fc4WHUpZmQYmzoQbit28qJhLjScbAQWU/0/*)#ujr0xunp","devices": [{"type": "seedsigner", "label": "Single Seed Zone Testnet"}]}'
E        ^
E   SyntaxError: invalid decimal literal
========================================================== warnings summary ===========================================================
src/seedsigner/views/menu_view.py:274
  /home/pi/seedsigner/src/seedsigner/views/menu_view.py:274: DeprecationWarning: invalid escape sequence \d
    return int(re.search("#(\d+)", lines[self.selected_menu_num-1], re.IGNORECASE).group(1))

-- Docs: https://docs.pytest.org/en/stable/warnings.html
======================================================= short test summary info =======================================================
ERROR tests/test_decodepsbtqr.py
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 1 error during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
===================================================== 1 warning, 1 error in 1.47s =====================================================
```

nyxnor, 2022-02-19:
Stopped there as the tests failed and didn't try to make the signer work. Documented in case this facilitates the work of a seedsigner developer to make this work on bullseye.

closed 2022-02-19 by nyxnor