SeedSigner/seedsigner #150: 1 Million Sat Bounty for Touchscreen Demo

SeedSigner, 2022-02-19:
We are offering a 1 million sat bounty for a technical demo using this screen:

https://www.waveshare.com/2.8inch-DPI-LCD.htm

The demo should allow for direct pixel mapping/drawing using PIL/Pillow (Python Imaging Library) in the same way that we currently that library to render graphics to SeedSigner's screen. Please note that "mirroring" to this screen, similar to how a Pi desktop is displayed, or using a "windowing" solution won't suffice for our purposes -- we need to be able to directly manipulate pixels. I'd anticipate that this may require a custom driver / python library. The requested technical demo should also incorporate touch recognition.

Please feel free to ask any questions under this issue if anything isn't clear -- thank you!

mutatrum, 2022-03-01:
I have a basic test app working, which uses framebuffer. It's based on https://github.com/robertmuth/Pytorinox/blob/master/framebuffer.py, with a different pixel order. Would this be an acceptable route?

I haven't looked at touch interfacing yet.

SeedSigner, 2022-03-02:
I am most curious about the performance impact of the framebuffer on the Zero 1.3. I would assume it wouldn't be a problem on the Zero 2W, but it would be ideal if the 2W wasn't required to use this, or any other touch screen? I am guessing that framebuffer also opens up the possibility of supporting other screens, aside from this specific one?