SeedSigner/seedsigner #153: Seedsigner crashes if first seed is stored in slot 2 or 3

Caramba2Go, 2022-02-26:
Seedsigner (0.4.5) crashes if your first mnemonic seed is stored in slot 2 or slot 3 and if you then add a passphare to this seed.

100% repdoducable:

1. store seed to slot 2
2. add seed passphrase to slot 2

SeedSigner, 2022-02-26:
Really appreciate you documenting this here. 0.4.6 is likely the final release with the second version of our code base. 0.5.0 will be a complete rewrite, constituting the third version of our cod base, which should eliminate this bug. Thankfully it sounds like if you start with Slot 1, you can still add the passphrases to the other slots. Going to leave this issue open until 0.5.0 releases, thank you again.

kdmukai, 2022-05-01:
@newtonick this issue can be closed.

Caramba2Go, 2022-05-02:
Closed issue as it is fixed now.

closed 2022-05-02 by Caramba2Go