SeedSigner/seedsigner #157: QR brightness issue on the 0.5.0 Pre-release 1

mccolister1017, 2022-03-03:
can't scan anything on the rasp it's too shiny (full hd webcam)  got it working with some trick but i think it's a bug everything is ok with the main version.

or maybe you could add a ( Brightness Settings ).

SeedSigner, 2022-03-03:
Thank you for this, going to add brightness adjustment to XPUBs in addition to PSBTs. Thank you!

jdlcdl, 2022-11-21:
I suspect that this issue might deserve to be closed?  Reason being that I've never had a problem adjusting brightness while seedsigner is displaying a qrcode for my computer to read.  For me, it's absolutely necessary to dim the qrcode... and so far it has always worked well for me (but Im using webcams of 15-20years old).

Then again, re: "or maybe you could add a ( Brightness Settings )." reminds me of my ramblings on telegram a couple weeks back, where I was playing with rpi's pwm to power the backlight, and I was not wanting to do this for qrcode reading but rather for saving my own tired eyes in low-light.   If this is the reason why this 'bug' issue remains open, then I'd like to consider working on a "brightness" user settings change, but I'm not sure where that might fall on the priority list. 

I'll check back on this issue for feedback from others.. and of course I'll make note that it's my belief that seedsigner-os intends to disable rpi-pwm in the near future, because it would be silly to devote too much work to brightness control via pwm-backlight if pwm remains on the chopping block.

jdlcdl, 2022-11-22:
for future reference: related to my thoughts on "powering backlight via pwm to adjust screen brightness":

on my RPi2b: I see a +5-6% cpu-usage cost on the python process powering the backlight pin via pwm at 1000hz and 50% duty cycle... as opposed to just setting it high.