SeedSigner/seedsigner #166: Provide reproducible builds

Giszmo, 2022-03-25:
The [Manual Installation Instructions](https://github.com/SeedSigner/seedsigner/blob/main/docs/manual_installation.md) look far from deterministic and while #105 would probably help to get something more predictable, there is still many sources of randomness in the image creation process such as `sudo apt-get update && sudo apt-get install -y ...`. Without deterministic builds, users have to trust the provider of the image. While the seedsigner does support mitigations for many attacks:

* (forced?) dice ware vs. fake randomness
* air-gap vs. key leaks via radio devices
* third-party companion apps vs. receive-to-malicious-address

it feels wrong to assume the device can't scam its users. Or do providers claim just this?

newtonick, 2022-04-21:
I'm definitely interested in reproducible builds, but I'm not aware of any turn key solutions to make this happen. We are hoping to make steps in that direction, but it's going to take sometime. Any suggestions or open source projects you can point us to?

Giszmo, 2022-04-22:
Unfortunately I can't point you to anything useful but generally it might "just work" for a short amount of time, so it would be worth a try if doing the image on two machines yields anything comparable.

From there, you can always install specific versions instead of upgrading to whatever is the latest etc.

DesobedienteTecnologico, 2022-06-09:
Reproducible build available @Giszmo :
https://github.com/SeedSigner/seedsigner-os

StingJohnson, 2022-09-28:
the os i have built with https://github.com/SeedSigner/seedsigner-os doesn't meet the same sha256 from the official release. In fact, it is a lot smaller in terms of size. What could go wrong?

SeedSigner, 2022-09-28:
Right, these will never produce the same hash because the standard release has a much larger OS footprint than the build-root OS that you used. The build-root OS may end up being reproducible, we're not sure yet whether it's feasible/possible and how much work will potentially need to go into making it reproducible. Feel free to ask if you have any other questions or concerns.

Giszmo, 2022-09-28:
If the released binary can't be reproduced, what is reproducible then?

StingJohnson, 2022-09-29:
yes, i am thinking the same. Although it is definitely a handy device and seems legit, if it can't reproduce the released binary, it is missing the foundation on which all the trust are relying.

SeedSigner, 2022-09-30:
Some clarification is helpful here. The released binary based on the current conventional build will never be reproducible, never; it's just not possible given the process used to create the conventional build. The new SeedSigner-OS is a completely different approach to building the Linux OS upon which our python software operates, so it will never match the data in conventional build process, which utilizes a stock version of Raspberry Pi's pre-assembled OS. They are completely different animals and it's important to understand this concept as you attempt to evaluate our builds. That said, @DesobedienteTecnologico believes that a reproducible build is in fact possible with the BuildRoot OS that the new SeedSigner OS leverages, but it will take significant time and effort to get there. We have to figure out how to exclude the multitude of timestamps that buildroot incorporates into the OS so that the binaries can be properly compared and validated. I would really invite both of you to dig in and see if you can help; we only have a handful of generally uncompensated developers who work really hard on our project. If you lack the technical skills to meaningfully contribute to our development efforts and still would like to see our software continue to be developed and improved, I would encourage you to make a donation to our development fund, which is the primary way that we can motivate and compensate contributors for their work. Donations are accepted at: https://seedsigner.com/donate-now/

StingJohnson, 2022-10-25:
So, why building from the source (your seedsigner-os) is significantly smaller in size than the binary release? I forgot the exact number but i do remember that the one built from source is way lighter.
Now I can understand that due to the version of python and all that makes the hashes don't match. But what causes the huge difference in size?  Thanks!

SeedSigner, 2022-10-25:
The engineers working on SeedSigner OS have included only the necessary functions that our software requires, all of the additional functionality and code that the Pi Foundation includes in their stock builds has been removed. That is the reason why the binary is so much smaller.

StingJohnson, 2022-10-26:
If SS OS only includes the necessary functions, which is a good thing, then why making the stock builds that bloated. Shouldn't stock build include ony the necessary fucntions? 
We need it only to be as a Seedsigner not anything else, don't we?

SeedSigner, 2022-10-26:
We have just gotten the custom OS to the point where an experimental release made sense. As it is further tested and refined, the plan is to phase out the stock builds, but we're extremely risk averse and want our prepared images to be stable and reliable for our users. We are a project composed of volunteer developers with very little in the way of funding; we will get there but it may not be as quickly as you would like.

moonsettler, 2023-11-21:
shouldn't this issue be marked as resolved now?

ktecho, 2025-07-23:
Latest releases are reproducible, so yes, it should be closed:

https://walletscrutiny.com/hardware/seedsigner/

(to get more details, click on any of the specific verifications)

closed 2025-07-25 by Giszmo