SeedSigner/seedsigner #169: Feature: Coin flipping with QR-code as starting point

Giszmo, 2022-03-30:
Apologies if I miss something. I never touched a SeedSigner yet but to my understanding, the device **requires** the user to 

* print 2048 words
* pull 11 words out of a bag
* ask SeedSigner for the 12th word (which currently wastes some entropy according to #132)
* use SeedSigner to draw a QR-Code for quick input

I'm not sure about the encoding of the QR-Code but it might allow to start with the QR-Code and coin flips, skipping the printer part:

* draw empty QR
* throw coin and skip a square for every heads and paint the tails-squares black
* ask SeedSigner how to paint the final n squares to make checksum right
* ask SeedSigner for the corresponding words

SeedSigner, 2022-04-04:
For coin flips, this is the implementation I've seen that would seem to make the most sense:

https://twitter.com/fgbernal/status/1502300290971807757?s=20&t=ebtl8DitMd9XALySokJJiA

Hopefully now that we're close to having 0.5.0 we can get Fran to put his work into a PR.   : )

kdmukai, 2022-05-25:
I don't think there's any way to implement this as this issue is written. @Giszmo is talking about interactively constructing a QR code from sequential random inputs. QR codes can't be determined block-by-block in a linear construction process.

What @SeedSigner links to is a separate concept: Reading a coinflip chart as an easier way to input entropy.

Recommend this issue be closed @newtonick.

Giszmo, 2022-05-25:
Sadly I have to agree with kdmukal. Closing.

closed 2022-05-25 by Giszmo