SeedSigner/seedsigner #184: Add a PayNym to "Donate" screen

punterwantsawhalepass, 2022-04-18:
https://twitter.com/KeithMukai/status/1510358760891588611?s=20&t=pZoRnOGAgRe0MoCXOQf3Xw

newtonick, 2022-04-21:
@SeedSigner are you interested in having a PayNym for SeedSigner? It requires creating a BIP47 reusable payment code and then registering for a PayNym for the payment code on a Samourai hosted service https://paynym.is

The only coordinator I know of that supports both PayNym's and SeedSigner is Sparrow Wallet.

I'm pretty happy with http://seedsigner.com/donate as a means for private donations via lightning.

punterwantsawhalepass, 2022-04-21:
maybe a add a QR that opens https://seedsigner.com/donate-now/

right now "seedsigner.com" is what's on the Donate screen

SeedSigner, 2022-04-21:
I'm open to a paynym but would need to dig in to make that happen.

I do believe that @kdmukai had planned to implement a QR code that linked to the donation page, just a lot going on and I don't think he has gotten to it yet.