SeedSigner/seedsigner #189: Edit suggestion for - "away" from all online devices

punterwantsawhalepass, 2022-04-19:
may need to change that, away made me think of distance

"You must keep your seed words private and safe. Never store your seed words on an online device."

newtonick, 2022-04-21:
Issue is referencing this line of text/code: https://github.com/SeedSigner/seedsigner/blob/5e007e501044f518f8d3c8ab9505901cfba249bb/src/seedsigner/views/seed_views.py#L768

punterwantsawhalepass, 2022-04-21:
this might also be good language. 

"You must NEVER allow your Bitcoin seed words to come into contact with the Internet."

source: https://observablehq.com/@jimbojw/grokking-bip39

SeedSigner, 2022-04-22:
Just taking a stab -- "Never input your seed phrase into a device that connects to the internet."

newtonick, 2023-01-27:
> Just taking a stab -- "Never input your seed phrase into a device that connects to the internet."

+1 for this option if just trying to resolve the immediate issue

kdmukai, 2023-01-27:
Photos of proposed updated screens would help here (or rather in a PR). Sometimes messages just don't look/read quite right onscreen.

jdlcdl, 2023-01-27:
> Photos of proposed updated screens would help here (or rather in a PR). Sometimes messages just don't look/read quite right onscreen.

I agree, especially since it's not obvious where different phrases will wrap.  Back here shortly with some images of all the proposed phrases in this thread... with the help of "enteropositivo/seedsigner_emulator".

jdlcdl, 2023-01-27:
Thanks to the work in enteropositivo/seedsigner_emulator, this was easy to do from my desktop.

Once we decide on one of these, I'll be happy to verify that it indeed looks the same on the seedsigner.
If anyone would like to propose changes, just leave a note here, I'd be happy to get a screenshot while this procedure is fresh in my head.

Currently:
[image]

Proposal A and B, thanks to @punterwantsawhalepass 
[image]
[image]

Proposal C, thanks to @SeedSigner (w/ a +1 from @newtonick)
[image]

jdlcdl, 2023-01-27:
I prefer Proposal C, because:
* it preserves vertical space around the message,
* it's an assertive command instead of really good advice,
* As a a rare "honorable usage of fear" as a motivator, it reminds the user that they've done it wrong for years, and to take action immediately.

jdlcdl, 2023-01-27:
ATTENTION: I had hoped that the SeedSigner Emulator would accurately represent where text would wrap, but this is not the case everytime. :(

I'm going to issue a pull request for this, leaning towards Proposal C and carrying similar text into the SeedQR warning.  I'll take photos of what it really looks like on SeedSigner hardware.

SeedSigner, 2023-01-29:
Thank you for your work on generating these previews.

bitcoinprecept, 2023-05-09:
Some better late than never thoughts on the topic here... (without thinking through any specific proposals to the outstanding edits proposed above)

Some people may not fully think through what it means to interact with "devices that connect to the internet" (or however it's phrased in the multiple variations above). I think it's worth offering users a clear warning to not interact with their seedphrase data within site of webcams of their cell phone camera. This may be an edge case not worth solving for within the limited Seedsigner UI, but it's worth considering that your camera could be a vector to spy on you/steal your data even if you don't voluntarily "take" a photo with it.

The best practice, in my opinion, is to be fully away from internet connected devices and cameras when interacting with seedphrase data. If it were possible to relay this concept to users, that would be my vote.

PR #314 (2023-07-20): seed backup warnings: text changes
resolves #189

In the original issue, I had previously shared "screenshots" from SeedSigner Emulator which are close -- but not exact representations of what these text changes will look like on the seedsigner.  In a few cases, the text wrapping is NOT representative, so I've taken photos of "currently" and "proposed" which I'll include here.

Seed phrase warning screen currently:
[image]

... and proposed:
[image]

I've carried similar wording to the related SeedQR warning screen, currently:
[image]

... and proposed:
[image]
closed 2023-07-20 by newtonick