SeedSigner/seedsigner #200: Question regarding LCD HAT

AngeloMetal, 2022-04-30:
Hi, in my country there's no ``Waveshare 1.3inch LCD HAT 240x240``, which is required for the SeedSigner's setup. I'm a newbie when it comes to RPi, so I really need your help.

Could I purchase the [Adafruit 1.54" 240x240](https://www.hellasdigital.gr/electronics/displays/adafruit-1.54-240x240-wide-angle-tft-lcd-display-with-microsd-st7789/) instead? I know the pixel count matters, but I was wondering if it has to be specifically a Waveshare.

SeedSigner, 2022-05-01:
Hi there. So the display itself will work fine, but the challenge with that one is that you will then need to implement the rest of the controls via other components. Is international shipping into your country an option? Waveshare sells the hats directly to customers (see: https://www.waveshare.com/1.3inch-lcd-hat.htm) or I have also had good luck sourcing through Ali Express sellers (such as this one: https://a.aliexpress.com/_mNNgxxG).

AngeloMetal, 2022-05-01:
Hi, I think I'll just buy it from the aliexpress, thanks!

closed 2022-05-01 by AngeloMetal