SeedSigner/seedsigner #205: [Enhancement

bitcoinheiro, 2022-05-09:


kdmukai, 2022-05-09:
If I understand the flow correctly, there probably wasn't any point in doing your change address verification this way.

The idea is that you should already have a copy of your multisig wallet descriptor in a protected, offline format (e.g. printed out when you created the multisig wallet and kept in a safe place). This way if your wallet software was later compromised and started sending you evil PSBTs that try to steal your change, scanning the original multisig wallet descriptor would catch that deception.

But if you're just scanning the live multisig wallet descriptor from the software that constructed the psbt, your "source of truth" is the potentially-compromised software. If it can lie to you in the PSBT it creates, it can lie to you by giving you an altered multisig wallet descriptor. So in that flow you may as well just trust the PSBT as it was constructed since the verification step doesn't add any further assurances.

That all being said, I think the reason that we don't keep the multisig wallet descriptor in memory is that it creates an awkward UI if one is already in memory. How do you refer to it? There isn't a single fingerprint identifier like there is for a standalone seed. And we can't just assume that the one in memory is the one you want to use to verify your multisig PSBT so we'd have to refer to it somehow: "Use [____]" or "Scan Different Descriptor".

bitcoinheiro, 2022-05-09:
Understood. Yes, makes sense to just print the descriptor, problem is that BlueWallet (for example) does not offer to print the descriptor with QR codes. I can share it, export it, but as a TXT file with descriptor info in text format.

kdmukai, 2022-05-09:
See: https://twitter.com/KeithMukai/status/1506726109567672321

bitcoinheiro, 2022-05-17:
Perfect, added a section on how to convert the descriptor into QR code to the tutorial.
I remember that you also had a companion software in development for the SeedQR generation, right?
Maybe this could also be there

closed 2022-05-17 by bitcoinheiro