SeedSigner/seedsigner #211: [Enhancement] Insert seed with BIP-39 index numbers instead of words

bitcoinheiro, 2022-05-25:
There are many seed backup solutions around nowadays that use the BIP-39 words index (digits from 1 to 2048) to back up the seed. Recently I also reviewed the "Krux" device and they have the option to import the seed into the device using only the numbers index instead of writing the words. I think if might be a good idea to add this for easy input with numbers.

Example from Krux:
https://github.com/selfcustody/krux/blob/main/src/krux/pages/login.py
Look for "load_key_from_digits"

SeedSigner, 2022-05-25:
I am guessing this made sense for Krux because of hardware input limitations. The downside to this approach is that while BIP39 words have built-in context and meaning for users (humans think predominantly in words), numbers inherently have less context and would seemingly be more error prone (though of course the final word would still be an error flag). That said, I am not be opposed to incorporating this option -- I would be curious to get thoughts from @kdmukai on upside/downside and what workflow / menu options might look like.

revetuzo, 2022-05-25:
Basically this is what the SeedQR does, which seems to evolve into something like a standard now with Sparrow Wallet supporting it too. I see three problems here for manual interaction:

a) Is it 1 to 2048 or 0 to 2047? Or implement both and choose?
b) Which language? English only?
c) Why would you do that? Defeats the purpose of the mnemonic. Here's what BIP-39 says:

> A mnemonic code or sentence is superior for human interaction compared to the handling of raw binary or hexadecimal representations of a wallet seed. The sentence could be written on paper or spoken over the telephone.

> This guide is meant to be a way to transport computer-generated randomness with a human-readable transcription.

kdmukai, 2022-05-25:
They handle the `word_num - 1` problem, but I hate that they assume the user is entering indices from a list that starts from 1.

BIP-39 has a trivial yet nightmarish problem with these index numbers. Exacerbated by Github displaying the wordlist file with usual line numbers (starting at 1) and even backup products using 1-based indices in the lists they distribute with their products.

I don't think it's a bad idea to add index number entry, but I would find it very hard to stomach incorrect 1-based index numbers. This is probably a losing battle but I'd rather keep educating people that the actual technical spec starts the indices at 0.

And now that I think about it, if this isn't implemented, there is already a simple workaround: the human looks up the wordlist on their own. I don't think the slight extra convenience is worth having this brought into the UI. Optionality is always good, but only if it solves an otherwise thorny or impossible problem for the user.

So I'd say NACK.

bitcoinprecept, 2023-05-09:
If you already have your index numbers, it doesn't seem like much work to reference the official BIP-39 wordlist online and enter the words corresponding to the index numbers in your seed documentation.

In a vacuum I like the option of being able to manually enter your seed word index numbers as proposed here (more options are better), but the potential user confusion around whether your numbers start from 0 or 1 is probably a bigger stumbling block than simply expecting that user to reference the word list to enter words.

jdlcdl, 2024-08-05:
Someone spoke of this today in the telegram group.

A thought on ux for this feature:
* It matches the line numbers for the bip39 words list file in github.  (This used to make me cringe.  But recently I've flipped a bit in my brain that says: `"Numbers" are for humans, and they start at 1.  "Index" or "Indexes" are for programmers they start at 0.`, and since then, it bothers me less.
* A 10 digit number pad is easier to traverse than a 26 letter keyboard.
* There are more chances to short-circuit to the correct word:  User hits enter as soon as the number is expressed (could be 1, 2, 3, only half the time will it be 4.  When the first digit is 3 or higher, it cannot be more than 3 digits total; when the first digit is 2 and the second digit is 1 or higher, same.
* It's yet another way to backup a bip39 mnemonic that is arguably easier to conceal and/or to hide in plain sight?

nomadbit21, 2026-10-01:
Opened a PR that adds loading (and backup) a seed from BIP-39 word indexes (1–2048), plus the matching 12-bit row for punch-plate backups.

Tools → Seed Word Indexes → Numbers or Binary → 12 or 24 words. The last entry uses the existing finalize / invalid-seed path. English BIP-39 only.

[PR_URL](https://github.com/SeedSigner/seedsigner/pull/1051)