SeedSigner/seedsigner #212: Improve/Rebuild the SeedQR templates as SVGs

kdmukai, 2022-05-25:
The original/current implementation of the [SeedQR templates](/docs/seed_qr/printable_templates) are built as clunky html table elements.

Due to html/css weirdness (more like: my own failures in tweaking them), the wordlist card generally doesn't print at the same scale as the QR code grid.

The QR code grid (and maybe the wordlist card?) should be rebuilt as probably an SVG. Should be mostly straightforward to alter the [template_generator.py](/docs/seed_qr/printable_templates/template_generator.py) to create the SVGs.

Someone with only a modest level of python + knowledge of SVGs, modest html/css should be able to take this on.

@newtonick, do we have a "Good first issue" tag?

EverydayBitcoiner, 2022-12-24:
I've done some work generating qr codes directly to pdf and would be happy to make some adjustments. Here is what I've done for multisig backup stuff https://github.com/EverydayBitcoiner/multisig-backup. It is based off of SeedSigners baseball card design and SeedHammers multisig backup plates, but I can change it around to look like whatever we want. Is this generally just for people to print and use?

[image]

kdmukai, 2022-12-24:
I think we'd want a variety of template options. Basically like the open source enclosures; people can peruse what's available and use or modify whichever.

I realized that I could quickly create templates just via Google Sheets rather than wrestling with CSS, SVGs, etc.

And I like the idea of having some totally bare-bones options.
* Just the basic QR code on its own.
* Just a basic wordlist. 
* Simple combo of the two.
* And then more stylized options like the baseball card design(s).
* etc.

---

I've been meaning to PR these in. I don't think I'd consider them final yet, but you get the idea.

[SeedQR Templates - 21x21 dots.pdf](https://github.com/SeedSigner/seedsigner/files/10298666/SeedQR.Templates.-.21x21.dots.pdf)
[SeedQR Templates - 21x21 grid.pdf](https://github.com/SeedSigner/seedsigner/files/10298667/SeedQR.Templates.-.21x21.grid.pdf)
[SeedQR Templates - 25x25 dots.pdf](https://github.com/SeedSigner/seedsigner/files/10298668/SeedQR.Templates.-.25x25.dots.pdf)
[SeedQR Templates - 25x25 grid.pdf](https://github.com/SeedSigner/seedsigner/files/10298669/SeedQR.Templates.-.25x25.grid.pdf)

EverydayBitcoiner, 2022-12-24:
Sounds good. I'll mess around a little a make some different options.

krishnavbajoria02, 2025-04-01:
I’ve been reviewing this issue and trying to understand its current status. I noticed that many of the links mentioned in the conversation (e.g., SeedQR templates and related branches) lead to 404 errors, which makes it difficult to fully assess the work that has already been done. 

> I've been meaning to PR these in. I don't think I'd consider them final yet, but you get the idea.
>
> [SeedQR Templates - 21x21 [...]

Additionally, upon checking the repository, I see that these templates seem to already exist under ```docs/seed_qr/printable_templates```.

Since I’m new to contributing here, I’d appreciate any clarification or guidance on what’s left to be done for this issue. If SVG work is still needed, I’d be happy to help with the conversions or implement additional features as required.