SeedSigner/seedsigner #213: Export XPUB as plain-text

revetuzo, 2022-05-25:
Hi all,
plaing with Samourai Sentinel I realized that it only accepts plain-text x/y/zpubs. The closest to this is the BlueWallet export, but the fingerprint and derivation part is not understood by Sentinel. How about adding a simple plain-text static QR export?
Cheers!

kdmukai, 2022-05-25:
BTCPay server also had a problem with the `[fingerprint/<derivation_path]xpub...` format. But I believe that's been resolved. I think all the other coordinators we've tested (Specter, Sparrow, BlueWallet, Nunchuk) all accept this format.

So it might be better to prod the Samourai team to broaden their support to read and discard the prepended info to get them in line with what otherwise seems to be a consensus standard.

oren-z0, 2026-08-15:
See PR at: https://github.com/SeedSigner/seedsigner/pull/1007