SeedSigner/seedsigner #216: Feature Request: Generate & Display Receive Addresses from Seed / Wallet Descriptor

SeedSigner, 2022-05-26:
Intending this initially as a feature for single-signature wallets to keep things simple and see if there is value there for users.

The idea is to take an in-memory seed and let users generate a wallet and receive addresses so as to be able to receive funds without ever having even exported an XPUB for initial wallet configuration, for an increased level of privacy / secrecy (the coldest of cold storage). After entering the wallet configuration details, users could display receive addresses (ordered by their index number) and of course corresponding QR codes to make a deposits. Could also be interesting if users could generate a QR-encoded wallet descriptor to easily import the wallet into a coordinator when they were ready to spend. This feels like a feature for more "advanced" users, would possibly want this functionality obscured or disabled by default?

Am curious if others have any opinions on whether this concept is useful, or even a good idea to consider implementing.

EverydayBitcoiner, 2022-12-27:
It looks like the dev branch has implemented the address explorer for in-memory seeds and multisig descriptors, but does not support single sig descriptors or xpubs. Is this something that is being actively worked on somewhere? Is this something that we want to implement? This seems like something that would fall under issue #126.

Should I make a seperate issue for the multisig descriptor output? Is that something we want to keep on our to do list?

For the wallet descriptor QR export, for single sig what is the difference between the xpub export and the descriptor export? Does the descriptor export just include the script type?

Jaffasoft, 2025-03-17:
> Intending this initially as a feature for single-signature wallets to keep things simple and see if there is value there for users.
>
> The idea is to take an in-memory seed and let users generate a [...]

Can this be done for Multisig too? 

I'm not sure if i do it wrong being new to using SS, but when i create a Multisig i have to export each seed xpub separately create an online Sparrow Wallet and import the descriptor into SS to see an address etc. Why not have seed signer just be able to scan a few seed QR codes and tick to select to create Multisig descriptor and addresses? Then all the things described above for single sig and privacy ect can be done completely offline with SeedSigner and nothing else. 

Maybe i'm overlooking something and it can be done?