SeedSigner/seedsigner #219: Exception scanning compact QR

enteropositivo, 2022-06-01:
### Current situation:

Now I can emulate seedsigner on desktop,  webcam working too: 
https://github.com/enteropositivo/seedsigner-emulator

And I can scan a  standar seed succesfully

### But I have a problem scanning compact seeds  

Using the sample QR from Seedsigner repo  
[https://github.com/SeedSigner/seedsigner/blob/dev/docs/seed_qr/img/compact_12word.png](https://github.com/SeedSigner/seedsigner/blob/dev/docs/seed_qr/img/compact_12word.png)

I got the following exception  on line 400 when code does **_bin(b)_**:

`TypeError("'str' object cannot be interpreted as an integer")`

[image]

**Some Notes** 

- That the same seed works OK for me using  SeedSigner hardware on Rpi Zero, but the same code doesn't work on a desktop enviroment
- The code is unchanged from seedsigner repo ( I only change display driver under /hardware/ folder )
- May be a problem with python ?  (I'm using v3.10)

SeedSigner, 2022-06-02:
Will let @kdmukai chime in, but I believe this is because we implement a specialized QR code library to read the compact SeedQRs.

enteropositivo, 2022-06-05:
### SOLVED

I found a solution.

The libraries that zbar downloads on Windows enviroment are very old, "SeedSigner requires zbar at 0.23.x or higher."

This is the code of **build.sh** from **pyzbar**  at line:9  that downloads  v0.1 of libraries

```sh 
curl --location --remote-name-all "https://github.com/NaturalHistoryMuseum/barcode-reader-dlls/releases/download/0.1/{libzbar-32.dll,libiconv-2.dll,libzbar-64.dll,libiconv.dll}"`
```

### To be able to use compact QRcode  on Windows ...
we need to install the newest zbar libraries this way

Install MSYS2 building platform for windows
[https://www.msys2.org/](https://www.msys2.org/)

Install Zbar libraries and dependencies package
[https://packages.msys2.org/package/mingw-w64-x86_64-zbar?repo=mingw64](https://packages.msys2.org/package/mingw-w64-x86_64-zbar?repo=mingw64)

**Now pyzbar wrapper needs to load libzbar-0.dll v0.23.94 library**
So edit:  src/pyzbar/pyzbar/zbar_library.py   

Change line 23 and 24  as
```python
if sys.maxsize > 2**32:
        # 64-bit
        fname = 'libzbar-0.dll'
        dependencies = []
```

Change line 58 as  (path of MSYS2 Zbar libraries)
```python
  dependencies, libzbar = load_objects(Path('C:\\msys64\\mingw64\\bin'))
```

And NOW the emulator works with all the functions as on Raspberry Pi Zero

kdmukai, 2022-06-05:
Amazing bit of tech sleuthing, @enteropositivo!

Looking forward to seeing where this work goes! That being said, I would guess that Windows would be a kind of last-resort OS option for an emulator. Something like Tails or some other Linux flavor would be safer.

SeedSigner, 2022-06-06:
Agreed, this is great work @enteropositivo! The Windows implementation has a strong use case for tinkerers and development, but agree with Keith that Tails is important too. It looks like you have begun TailsOS instructions in the readme?

enteropositivo, 2022-06-07:
The main objective of the emulator is to be able to modify the code or play with it in a simple way without having to use the Raspberry Pi.  

But it can also serve as a tool to calculate the last word of the seed or even use it in Tails OS (currently works)
Anyway I still have to do more tests with the camera since in Ubuntu it goes very well but in Tails sometimes it has problems.

The purpose of making it compatible with Windows is basically for curious people who want to try it on testnet before deciding to buy a Raspberry (of course, if they can find one because there is no stock  :) )