SeedSigner/seedsigner #231: [feature] USB Smartcard reader support

openoms, 2022-07-09:
There are only so many Specter Shileds around, but storing secrets on smartcards become my favourite method.
Consider the convenience of the cheap, PIN protected, encrypted storage which is very well suited for travel (especially compared to plaintext seed words or QRcodes).

Since the SeedSigner currently runs a full linux distro the driver support should be fairly straightforward through USB.

The setup in mind:
* the SeedSigner could unlock with the PIN and read the secret from the smartcard - same as the Specter Shield
* could even be able to initialize the cards (reset and upload the MemoryCardApplet.cap)
[image]

Parts:
Cards: https://www.alibaba.com/product-detail/JCOP-Dual-Interface-Support-RSA4096-ECC_1600070838098.html

The reader pictured: 
ACS ACR39U-N1 PocketMate II USB Smart Card Reader
https://www.amazon.co.uk/dp/B0758TS5JR/

Notes to install the card reader an initialize on Debian/Ubuntu: https://gist.github.com/openoms/510b2876cab19e15c4190456ea8aad82

Code in Specter DIY: https://github.com/cryptoadvance/specter-diy/blob/master/src/keystore/memorycard.py
Issue about supporting other USB or hard-wired card readers: https://github.com/cryptoadvance/specter-javacard/issues/1

SeedSigner, 2022-07-10:
It's an interesting concept though I don't have any practical experience in working with smart-cards. Do you have any information on the data-storage expected lifespan of the smart cards? Are there incidents where they can malfunction or their stored data becomes corrupted or otherwise inaccessible? I'm leery of recommending users store PKs in digital format without some kind of secondary backup (usually an analog one). But if there is good information on their reliability / durability, and the quorum is large enough, I could potentially see using them as a primary / solitary PK storage mechanism. Also open to any thoughts that @kdmukai or @newtonick may have.

openoms, 2022-07-10:
However smartcards are generally used widely for years without any corruption issue (doesn't even compare with the magstripes on bank cards) agree that any digital media shouldn't be the only, non redundant backup.
They are basically disposable secure chips which lock permanently after 10 failed unlock attempts (as configured for the Specter).

openoms, 2022-12-13:
> A smart card reader used like this would have to be dedicated, to maintain the air gap. I was reading a fascinating article by Brian Krebs on how the USA military ended up having .mil users buy their own insecure readers to be used (with malware infected drivers). Awful situation, but it highlights that the card reader can arrive pre-infected. (not the card is infected necessarily).
> 
> The proliferation of .mil service members buying MITM devices unwittingly/NMP is awful to see. thinking
> 
> So I would worry about that angle. Thoughts? M

All components can be added to the threat model. It is about optionality and I think a dedicated smart card reader (just as a dedicated Rpi0, sceen, camera and SDcard(s)) is a realistic requirement.

The advantage to have a general purpose, unbranded, but PIN protected and encrypted card holding a seed versus a plain text QRcode or seed backup is huge.

SeedSigner, 2022-12-14:
Still curious to hear any thoughts that @kdmukai or @newtonick may have?

newtonick, 2022-12-14:
I'm not against the seedsigner project exploring this idea. I agree with @openoms about adding optionality. DIY signing devices are a perfect place for folks to explorer different solutions for different threat models. I personally have no experience with smartcards and won't likely spend the time to explore this rabbit hole in the near future. Adding support for a smart card reading in SeedSigner OS will likely be more time consuming than adding it to the Raspberry Pi OS.

I'd love to see someone do a POC via manual build steps. This would help show what development steps would be required and gauge seedsigner user interest.

3rdIteration, 2023-09-23:
Is anyone here still interested in this? I've been playing around with the satochip and seedkeeper  smartcard based devices and adding an integration that allowed both seed storage and also on-card signing would actually be trivial, as both the java applets and a python API is already available. (It would simply be a matter of adding the UX elements)

I'm happy to knock up a proof of concept if there is any interest.

jdlcdl, 2023-09-24:
> ...a proof of concept if there is any interest.

I will certainly follow any work you do on this, however it feels not so seedsigner-ish, at least to me, for my following opinion:

I've come to enjoy the half-duplex control that I have when transmission of information is via qrcode.  I get to control who is talking and who is listening... and if I want to, I can put another airgapped-or-not device in between to make sure that what is being said is what I want to be said... without the other listener hearing.

Do I believe that normal people would do this?  Of course not... but that I CAN is what matters to me... so that I can do this now and again, and so that the rest of the time I know that I'm still controlling who is talking and who is listening.

With electronic transmission, wifi/bluetooth/usb, the user never knows what is being said, whether or not there was two way communications, or the amount of information transmitted.

Still, if you work on this, because the tech interests me... you will certainly have my interest and my feedback.

kdmukai, 2023-09-24:
@3rdIteration an on-card signing use case wouldn't make any sense for SeedSigner. TapSigner and anything of similar functionality exists as a parallel hww-lite alternative.

As @Marc-Gee noted, back-dooring a smart card reader makes much more sense as an attack vector than trying to back door a Raspi Zero that might end up just controlling someone's Christmas lights or an NES emulator or who knows what. We'd be stepping into a much more aggressive adversarial arena. I'm not convinced the benefits are worth it.

@openoms:
> The advantage to have a general purpose, unbranded, but PIN protected and encrypted card holding a seed versus a plain text QRcode or seed backup is huge.

I don't deny that there is a use case, but overall I think the risk of plaintext mnemonic backup or SeedQR is strictly tied to the problems inherent with single sig. Yes, lots of people use single sig (and, of course, it makes sense as an initial destination when first taking self-custody), but all of the single sig security solutions just feel like band-aids. There is a spectrum, X is better than Y, but I just don't think that even Y is totally sufficient. So rather than trying to incorporate incrementally better Ys (that still aren't sufficient), I think the main focus should always be on encouraging multisig and making it as easy as possible.

As a tiny team of volunteers, @newtonick has the right take.

3rdIteration, 2023-09-24:
I'll finish writing a cli for this and knock up a basic PoC that just does storage with a USB connected, as this is trivial to implement. (Implementation via GPIO could come later if there is interest)

3rdIteration, 2023-11-30:
Demo code is here: https://github.com/3rdIteration/seedsigner

Video demo here: https://youtu.be/WHVWqJJBNdA

I'll post a guide on how to install from the "Manual Installation" environment, or if people prefer, can post a MicroSD image. 

Everything is just running nicely on x86

openoms, 2023-12-05:
Update:
I have managed to install and use the Satochip applet on the same javacard which was used for the SpecterDIY. They also have different, independent PIN codes.
See my notes here for Linux (started for Specter, now extended): https://gist.github.com/openoms/510b2876cab19e15c4190456ea8aad82

Started to test the setup with Sparrow Wallet by initializing the card, storing a seed and changing the PIN right from the GUI there.

Doing the initialization and signing on a self-built, stateless SeedSigner will be such a good use case, thank you very much for sharing this @3rdIteration

3rdIteration, 2023-12-23:
Added an updated version of the code here: https://github.com/3rdIteration/seedsigner/releases/tag/0.7.0%2BSatochip-Alpha1

Changes from previous alpha and some cases to 3d print are also included in the release info.

3rdIteration, 2024-02-27:
Just a heads up that I have pushed a more detailed video here:
https://www.youtube.com/watch?v=Rhs9z5uL7qg

And a proper beta release here: https://github.com/3rdIteration/seedsigner/releases (Running on buildroot now)

## Difference from Stock SeedSigner
* Multiple Smartcard interface options… 
   - Standard USB CCID/PCSC readers
   - PN532 NFC Reader Connected via I2C
   - USB Phoenix Type "Sim Reader" supported via OpenCT 
* Saving and Loading Seeds & Passphrases
   - Supports loading of Seed, Seed+Passphrase in one go, or loading passphrase independently. (Potentially from a different SeedKeeper)
* Saving and Loading Multisig Descriptors 
   - Also changed default behavior to keep Descriptor loaded until manually cleared. (Including descriptor appearing in the Address Explorer when loaded)
   - Descriptors are split up into a template and xpubs before being saved to SeedKeeper.
* Saving and loading generic secrets to a Seedkeeper card
   - These secrets can be either viewed as text or displayed as a generic text QR code.
* Initialising Satochip Cards with SeedSigner
  - Load any Seed from the SeedSigner on to the Satochip Card
  - Enable 2FA on the Satochip Card
* Flashing Satochip Applets to blank Javacards
   - Firmware comes bundled with applets for SeedSigner, Satochip and Satodime (Releases from Satochip Github)
   - Firmware is also bundled with the source code for all three projects, along with a modified varient for THD-89 based Javacards. (This can be built from source on-device at run-time)
* MicroSD Card Tools
   - Flashing MicroSD Cards with official SeedSigner Images (Bundled)
   - Verification of freshly flashed MicroSD cards against known images
   - Secure Wipe (Both with zeros and random data)
 
## Future Features & Improvements
* Add ability to lock/unlock/manage Javacards
* Add ability to view additional sttings/information for the Satochip cards
* Tidy up code and reduce re-use

3rdIteration, 2025-04-20:
Quick update on this. Basically I was getting a number of requests for folk to buy this as a kit, so decided to get something easier to assemble and more reliable than the PN532 clones which are all over the place.

I threw together a simple design based on the SEC1210 smartcard interface, which is supported in the Linux CCID driver. (Though needs some tweaks for the SEC1210URT variant) 

The prototype boards arrived and worked well, I'm currently getting second revision fabricated which makes some quality of life fixes and also adds a USB-C power input option so that the full sized reader obscuring the boards isn't a problem. 

It actually fits quite nicely in-between the existing display and board, so that is really nice.

I'm also thinking that I will get around to adding the functionality to have the Satochip card handle all of the signing on the card, meaning that the SeedSigner wouldn't actually have any secrets on it at all during normal operation, but rather acts as a mechanism to review the transaction details, load PSBTs, etc. (The idea of a small SeedSigner with a sim sized Satochip that handles the signing is quite compelling to me) Interestingly, this was actually one of the longer term goals for Specter DIY that never ended up being realised.

Images below.
[image]
[image]
[image]

I'll probably have these board generally available around mid May sometime, with the design being open so anyone can make them or have them fabbed.