SeedSigner/seedsigner #233: anti-exfil support

eliasnaur, 2022-07-23:
Is it feasible to implement anti-exfil ala Blockstream Jade? anti-exfil makes (near) impossible for a hardware signer to smuggle key material out through transaction nonces. See alse https://wally.readthedocs.io/en/latest/anti_exfil_protocol/

SeedSigner, 2022-07-25:
On its face, it would seem like a good idea. I am wondering if the best place to implement this would be within embit?

kdmukai, 2022-07-25:
In our use case what would serve as the "host" that is described in the docs? AFAIK the Jade has a unique dependency with an external server (something like Jade to online Green wallet that talks to Blockstream's infrastructure).

eliasnaur, 2022-07-26:
AFAIUI, the `host` is the wallet such as Sparrow.

kdmukai, 2022-09-09:
This would require support in `embit` and an extra round of QR exchanges with the coordinator.

The most likely implementation scenario would be to discuss this with [Stepan](https://github.com/stepansnigirev) to get his buy-in. Then we would:
* Implement support in `embit`
* Implement the extra QR exchanges as an option in SeedSigner and/or Specter-DIY.
* PR the extra QR exchanges as an option in Specter Desktop's signing flow.
* Collaborate with Craig Raw to get similar support in Sparrow.

But it's an all-or-nothing effort. None of these individual pieces accomplish anything if the other steps don't follow.