SeedSigner/seedsigner #234: Add extra 12-word (Compact) SeedQR test-vector to documentation

kravens, 2022-07-25:
On Specter DIY SeedQR imports we encountered an error for only some specific Compact SeedQR's, this is one of them:
[image]
> boy right bamboo major crew allow brass curious network giant awesome future

It might be worth adding as a test-vector so other implementations already test for whatever bug caused this:
https://github.com/cryptoadvance/specter-diy/issues/211 

SeedSigner Test-vector SeedQR's documentation I mention:
https://github.com/SeedSigner/seedsigner/blob/dev/docs/seed_qr/README.md#test-seedqrs

kdmukai, 2022-07-25:
Happy to add it, but I'd prefer if you can identify the exact bytes or byte sequence that tripped up the DIY first. Perhaps you could write a similar script [as I described here](https://github.com/cryptoadvance/specter-diy/issues/211#issuecomment-1194565350) to brute-force test lots of seeds and see which bytes any failures have in common.

stepansnigirev, 2022-08-13:
Makes sense to add test vectors that include `\r`, `\n` or `\r\n` as they are often used as end of line separators. In our case it was `\r`

kdmukai, 2022-08-14:
PR to update our docs with more test vectors is in. Thanks for identifying this issue @kravens and @stepansnigirev!

PR #241 (2022-08-19): Additional Compact SeedQR test vectors
Adds Compact SeedQR test vectors containing `\r`, `\n`, and `\r\n`.

Closes #234.
commit 9113eabe (2022-08-19): Additional Compact SeedQR test vectors

Closes #234.
closed 2022-08-19 by newtonick