SeedSigner/seedsigner #244: Have anyone tried to deploy Seed Signer to a Raspberry Pi Pico?

irae, 2022-08-16:
The Raspberry Pi Pico has a considerable processing power, as compared to Arduino and other microcrontrolers.

Some advantages would be having no linux, small power draw to run from batteries, smaller downloads, but more importantly, the Pico is about $4 USD instead of the $5 if you manage to find the Zero 1.3, or $15 if you go for the Zero 2 W. The downside might be harder to source camera for the Pico.

Just to illustrate, in some parts of the world it is difficult to source the Zero. I can't find the 1.3 here in Brazil or sellers that would ship it from China. Sellers from china do have the Zero 2 but it costs ~R$400 (around $79 USD), while I can find the Pico for ~R$35 + ~R$10 shipping (about $10 USD total).

So I researched a bit, found the [WaveShare Pico LCD-1.3](https://www.waveshare.com/wiki/Pico-LCD-1.3), which is ~R$40 and the [Arducam Pico supported cameras](https://www.arducam.com/raspberry-pi-pico-camera-modules/), which would be the most expensive hardware at $20~$40 USD. 

If I understood correctly, in theory other SPI cameras could work, some start at ~$3 USD. But I haven't found any camera that has supported drivers for the Pico, so I won't risk it, unless someone more experienced offers some guidance.

I already oredered the pieces and I intend to hack around it. While I wait about 20 days for the 3 parts to arrive, I was wondering...

Anyone has any insight on the idea? Or if you know of anyone that already tried that? If those more engaged with the project considered it and had a good reason to drop the idea? If Micro-Python can run the bitcoin libraries or I would need to go down to C (which I am less familiar) And if there is any top of your head advice about it? Any comment whatsoever would be appreciated. Thanks!

kdmukai, 2022-08-16:
I've been working on a custom MicroPython firmware, but mostly targeting the ESP32-S2. MicroPython is definitely SeedSigner's future and I think over time we'll be able to run on a broader array of hardware.

But the Raspi Pico / RP2040 only has 265kB of RAM that can't be expanded. That would make it very difficult for us to use for things like live video scanning for QR cods.

You can see most of my MicroPython + ESP32 + secp256k1 progress here: https://github.com/kdmukai/micropython-esp32

I do include some notes for compiling the custom firmware for RP2040 and I have tested it on a few.

newtonick, 2022-08-19:
Question answered, closing issue.

closed 2022-08-19 by newtonick

ffrediani, 2024-05-18:
@kdmukai @irae just added a Feature Request for ESP32 with some details and options. Please check out and comment if you have anything to add - https://github.com/SeedSigner/seedsigner/issues/560

habakan, 2026-11-09:
I tried this on a Pico 2 (RP2350, 520 KB SRAM, no PSRAM) with MicroPython.
SeedSigner's controller, views and models run, and a PSBT can be scanned, reviewed and signed.

It needed a small PIL for MicroPython, and embit ran out of heap, so I replaced it with bitwasm-signer, a signer I'm writing (C, with the PSBT parse in WebAssembly).
The heap is still tight, and the camera preview is greyscale.

Notes, if useful: https://github.com/habakan/seedsigner/blob/cefa651dc8a7f2d678f154255b5abdbaf23c1d36/pico/README.md