SeedSigner/seedsigner #247: FEATURE REQUEST: Usage of SS as a classic HWW - permanent storage of seed phrase(s)

WeAreAllHodlonaut, 2022-08-21:
Hello devs,

in comparison to classic HWW seed signer has unique features but it is not yet supporting to be used as a classic HWW.

In my opinion this would open up a greater usability and adoption since it could be used as a standalone HWW additionally.

Idea is just to have a setup phase (wallet generation) and a usage phase (signing txs) that is then protected by 4/5 digit PIN/passphrase you name it and stored encrypted permanently on the sd card.

In any case after PIN unlocking you could still WIPE the device and have the seedsigner interface from scratch. Or else swap out the microsd and flash with a new image. May be also handy for people that want to use or create microsd backups as additional safety net.

Hope you get the point and found the input useful for future advancement of the project.

BR

WeAreAllHodlonaut

SeedSigner, 2022-08-21:
I appreciate you taking the time to submit this feature request.

I would also like to get input from @newtonick & @kdmukai, but this sort of feature would need to be approached very thoughtfully.

The earliest I think we could reasonably explore this concept would be after our build root OS has become our primary release platform. But the PIN codes that are used on other hardware wallets aren't what is actually securing the private key, they are just what is being used to secure the device's interface, under the assumption that a manual brute-force attack on a hardware wallet would either be infeasible, or certain safeguards would be in place on the device whereby after a given number of failed attempts, the device wipes itself, or does some kind of factory reset to destroy the key.

Encrypting the private key on a loose MicroSD card is definitely possible, but the user would need to (and would want to) use a very strong passphrase to secure the seed/PK. Functionally speaking, we already offer this "feature" by virtue of supporting BIP39 word 13/25 passphrases, with the only difference being that users are required to secure their private key by writing the seed words down and/or transcribing a SeedQR (instead of saving the private key to a memory card).

I also think implementation of this kind of feature may create foot-guns for users, or give them a false sense of security were they to not use a sufficiently complex password that could be prone to a brute-force attack. The way our users currently interact with their seeds/PKs compels them to think carefully about their overall security posture, how they secure their seed/SeedQR, what opportunities for disclosure there may be, whether they choose a multi-signature setup over a single signature wallet, and whether they elect to use a BIP39 passphrase for one or all of their seeds.

This is just what comes to mind off of the top of my head, hopefully we get some impressions from devs or other people.

kdmukai, 2022-08-22:
I agree with @SeedSigner.

Mnemonic + BIP39 passphrase vs encrypted SD card offer roughly the same security tradeoffs.

One other minor concern: SD cards [aren't appropriate for long-term storage](https://www.integralmemory.com/faq1/how-long-will-data-stay-valid-for-on-a-memory-card-or-usb-flash-drive/). I also worry that they're VERY easy to lose or get mixed up with other cards if you're not careful. Someone had tweeted out that they dropped an SD card and it fell into a tiny crack in their floorboards!

So I think even with SD card backups it would be a best practice to still have a mnemonic or SeedQR backup (maybe in steel) somewhere.

If the goal is to have a more easily transportable backup that can escape a hostile border inspection, I think there are better options (e.g. [Border Wallets](https://www.borderwallets.com/).

If we didn't have SeedQR as a way to quickly load a seed, the SD card option would be more compelling to me.

So I'm mostly a NACK but if enough people seem to really want it, I'm still open to the idea.

This all being said, I think there is a strong use case to support importing the Coldcard encrypted SD backups. But for now I'd want it to only be one-way (i.e. implement import but don't implement exporting to an encrypted SD format).

WeAreAllHodlonaut, 2022-08-22:
Yes, this was my reasoning to generate additional copies on the device itself. less durable but smaller to store. Indeed flash has it's culprits...
 If it is to small, just use a SD adapter for storage?

So the idea was just to level up SeedSigner to have a similiar featureset as the Coldcard. Apart from this I am with you on not giving user a false sense of security. Despite that users still need to access their physical seed as SeedQR as raw information everytime you want to make a txs. Since the microSD is encrypted I see it as superior in this regard although you risk losing your funds in case you lose or do not secure the passphrase enough.

In general you could still just manually generate a seed using iancoleman and a seed backup on the same raspi with external monitor using another microSD and display, keyboard, mice and another linux image defeating the purpose of the seedsigner by itself. It is the UI  that seperates it from the rest but featurewise I would like to see it as close as possible to spread adoption.

bitcoinprecept, 2023-05-09:
I'm not part of the core team and do not speak for their perspectives, but I'll share my thoughts. TLDR very good and useful idea, but NACK

Optimizing for the core ideas of statelessness and air-gapped feel like the most important pillars of functionality to build around. This particular feature suggestion is good and has valid use cases, but IMO violates the core tenet of statelessness. 

When you try to be good at too many things, you end up not doing any of them as well as you otherwise could. I would argue the Seedsigner has developed a wholly unique vision and approach to Bitcoin key management (stateless air-gapped electronics) and it is far and away the market leader in this niche. If you really want coldcard feature set, users can go with coldcard who are the market leaders in that niche. Blockstream also has the Jade pushing into the market as a hybrid stateless/stateful option. I don't personally see a compelling reason for Seedsigner to try to implement that specific feature set under the project as presently described.

BenWestgate, 2024-09-06:
I had a thought about this and for very long lived seeds. Encrypting and saving the private keys involves trusting less hardware over time than statelessness. For example were your Pi to die, you would not expose seed material to a new camera and/or new input device again, just an encryption passphrase. It also creates an extra source of redundancy, the SD card AND the seed QR and is possibly more convenient, a microQR could be drawn in minutes to speed up future passphrase entry.

Electronic micro SD backups are the fastest to make and smallest. Waterproof and shockproof.

nkud90, 2025-04-15:
Let me share my vision of when statefulness in SeedSigner could become a valuable feature:

In a geographically distributed multi-sig setup, users are currently limited to making their seed backups publicly disclosable once someone gains access to them. While this risk is mitigated by the nature of multi-sig, it’s still worth exploring ways to make it even more secure.

By offering statefulness as an advanced, optional feature, we could enable such multi-sig setups to rely solely on encrypted, non-disclosable microSD cards stored in different locations. This approach would be a significant improvement over the current models, which involve either storing entire devices (stateful hardware wallets) or keeping seed phrases in a form that could potentially be disclosed.

kdmukai, 2025-04-16:
> In a geographically distributed multi-sig setup, users are currently limited to making their seed backups exposed once someone gains access to them.

bip39 passphrases can be used for multisig keys, too. Easy solution if you're worried about a key being "naked" out in the world.

BenWestgate, 2025-05-17:
He might have been considering the wallet descriptor for the multi-sig setups.  If the balance must remain private with 1+ locations compromised, that needs encryption.  I haven't seen any ideas for putting that to metal or paper with encryption, besides good old OpenPGP and printing a large QR of it.  The SD wins here for compactness.