SeedSigner/seedsigner #251: Include an incorrect Standard SeedQR test vector

moneymanolis, 2022-09-12:
While reviewing @kdmukai implementation of the Standard SeedQR format in Specter Desktop https://github.com/cryptoadvance/specter-desktop/pull/1870 and adding tests to it I realised that there only correct  Standard SeedQR test vectors on https://github.com/SeedSigner/seedsigner/blob/dev/docs/seed_qr/README.md. I think it would be good to include at least one with, say, a 4-digit number above 2047, to make testing of edge cases in the respective implementations more straightforward.

SeedSigner, 2022-09-18:
Seems reasonable to me but @kdmukai would have the best perspective.

moneymanolis, 2022-09-19:
Just fyi, I've included one pytest regarding this edge case in this commit: https://github.com/cryptoadvance/specter-desktop/pull/1870/commits/5a17507b3ec6ed8279f49ab96c5fd91c7e1433ba